esp8266/Arduino #2626: ESP12E with CP2102

vishalkothari, 2016-10-21:
Hello ESP8266 Arduino community,

My setup is as below
Hardware:			ESP-12E
USB to TTL: cp2102

Settings in IDE
Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Reset Method:		ck 
Port: 115200

With this setup when I plugin ESP-12E. AT commands are running properly.
First error I got was 
warning: espcomm_send_command: cant receive slip payload data

By looking at thread
https://github.com/esp8266/Arduino/issues/710 and 
http://arduino.stackexchange.com/questions/20219/upload-with-esptool-fails-with-espcomm-send-command-cant-receive-slip-payload

I downloaded new version of esptool.py and updated platform.txt

I am using Windows 7, so I was getting "error=193, %1 is not a valid Win32 application" when executing esptool.py.
To solve this, I used cx_freeze to generate esptool.exe

Now, when using this generated esptool.exe and trying to upload firmware, I get below error.
esptool.py v1.2-dev
Connecting...

A fatal error occurred: Failed to connect to ESP8266

AT firmware commands are still running properly.
Any pointers on how to solve this will be great help.

Regards,
Vishal Kothari

liquidfalcon, 2016-10-21:
What board are you using? Are you positive it's in UART download mode?

WereCatf, 2016-10-22:
@vishalkothari You need to pull GPIO0 LOW if you wish to flash new code on the ESP8266. If you are using a proper devboard there is probably already a button you can hold down when you reset the board or on Nodemcu and Wemos D1 you just need to select reset method in Arduino IDE as "Nodemcu" and it'll do it automatically, but you if you are trying to use a custom design of your own or just a bare ESP8266-module you need to pull the GPIO0 LOW manually yourself.

https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup

vishalkothari, 2016-10-27:
@liquidfalcon and @WereCatf . Thank you for your inputs.
There was a problem with my custom hardware which meant that GPIO0 wasn't getting pulled to LOW.

closed 2016-10-27 by vishalkothari