esp8266/Arduino #2627: Want to blink GPIO2 LED (built in) on ESP8266-01 (1024K) module

titusece, 2016-10-22:
Dear All,

I'm able to flash the empty arduino sketch and LED blink code (GPIO2) but not blinking the LED.
DO I need to flash any bootloader or anything ?
I have black color ESP8266-01 module.
While programming I pull down the GPIO0 pin and for booting I pulled up the GPIO0...
What could be the problem for not booting the ESP8266 blink code ?

This is the code I flashed into ESP8266-01 using Arduino IDE (Board selection is "Generic ESP8266")

const short int BUILTIN_LED1 = 2; //GPIO2
//const short int BUILTIN_LED2 = 0; //GPIO0
void setup() {
pinMode(BUILTIN_LED1, OUTPUT); // Initialize the BUILTIN_LED1 pin as an output
//pinMode(BUILTIN_LED2, OUTPUT); // Initialize the BUILTIN_LED2 pin as an output
}

void loop() {
digitalWrite(BUILTIN_LED1, LOW); // Turn the LED ON by making the voltage LOW 
//digitalWrite(BUILTIN_LED2, LOW); // Turn the LED ON by making the voltage LOW 
delay(5000); // Wait for a second
digitalWrite(BUILTIN_LED1, HIGH); // Turn the LED off by making the voltage HIGH
//digitalWrite(BUILTIN_LED2, HIGH); // Turn the LED off by making the voltage HIGH
delay(5000); // Wait for a seconds

}

Can anyone help me on this issue ?

2) I also want to perform some AT operation.
I tried to use this but not able to type or response from ESP8266-01 but junks.
https://github.com/espressif/ESP8266_AT

Can you also please help me to change the ESP8266-01 in AT mode ?

Regards,
Titus S.

WereCatf, 2016-10-22:
On the ESP-01 the blue, built-in LED is connected to GPIO1, not GPIO2, as far as I know.

Also, you cannot use both Arduino-core and AT-commands; it's one or the other, not both. When you flashed Arduino-code on your ESP it overwrote the AT-firmware, and if you flash AT-firmware on it it'll overwrite any Arduino-code.

titusece, 2016-10-22:
Thank you so much WereCatf.
I have seen the ESP-01 pinout and found GPIO0 and GPIO2, as you said GPIO1 is shared with U0TXD, checkout this link.
https://acrobotic.com/acr-00020
I am using the black colored ESP-01 module.
http://www.esp8266.nu/index.php/Tutorial_ESPEasy_Firmware_Upload

I have tried to use GPIO1 now but not blinking at all.
As of now, I am focusing to blink the LED using Arduino IDE.

Any suggestions/insights please.
Able to flash but not able to see the blink LED.
I appreciate your help on this,
Thanks again.

WereCatf, 2016-10-22:
If you have a USB serial-adapter connected to the ESP then it may be interfering with blinking the LED, I'm not sure. But yes, on an ESP-01 the LED should be on GPIO1, and yes, the same pin is also used by serial. It was moved to GPIO2 on newer modules, like e.g. the Nodemcus I have are based on ESP-12E and the LED is on GPIO2 there.

titusece, 2016-10-24:
Thanks WereCatf.
Yep! I'm able to see the on-board LED blinking.....thanks for the hint.
But I need to blink the LED using other GPIOs, is it possible? as I want to interface the DHT11 sensor and want to send using MQTT.
Also can you please provide any WiFi example which should connect my WiFi router and get the IP address for  ESP-01 module ?

WereCatf, 2016-10-24:
You can't transfer that built-in LED to any other GPIO, you need an external LED. With an external LED there is the problem that you should not be drawing more than max. 7mA current from the GPIO-pins, otherwise you may damage your ESP. A 330Ohm resistor in series with the LED should be enough, but the safest approach would be using the GPIO-pin to toggle a transistor and use the transistor to drive the LED.

As for examples: this Arduino-core for ESP comes with plenty of them. Fire up your Arduino IDE, then open the File-menu and Examples from there and just scroll down the list. I suggest perusing them, there are a lot of good ones there. Now, that said, here is a bare minimum sketch just to connect to WiFi:
``` cpp
#include <ESP8266WiFi.h>

const char* ssid     = "your-ssid";
const char* password = "your-password";

void setup() {
  Serial.begin(115200);
  delay(10);

  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA); //We don't want the ESP to act as an AP
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected.");  
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  delay(1);
}

```

titusece, 2016-10-24:
Superb!
Thank you so much again.
Let me try and update you.
I hope I can see the log in serial port.

WereCatf, 2016-10-24:
Oh, one thing: a lot of those examples don't use WiFi.mode(WIFI_STA)-call at all, which is something I consider a poor decision. The problem is, if you don't call WiFi.mode() when connecting to an existing AP the ESP will try to act as both an AP and a WiFi-client, and that can mess up other WiFi-devices on your network. Every now and then there's someone here with this issue, saying that when they turn on their ESP their phones or tablets can no longer access the Internet and stuff like that, and it turns out they aren't using WiFi.mode() to tell the ESP to only act as a WiFi-client, not AP.

I would recommend keeping this in mind, it may save you some gray hairs and frustration one day.

titusece, 2016-10-24:
Thanks WereCatf.

Can I use this code on ESP-01 and will it work & I could see the prints on serial port ?

void setup() {
 Serial.begin(115200);
 delay(10);
}
void loop() {
Serial.println("Hello world");  
}

This code is also not running and not even GPIO1 LED blinking now.
Not sure why :(     I spent so many hours to run one LED blink and WiFi code on ESP-01 module.
Any help greatly appreciated.
Thanks.

I have the ESP8266 (ESP-01, 1024K) black colored module.
Like below
http://www.esp8266.nu/index.php/Tutorial_ESPEasy_Firmware_Upload

Do we need to use particular Board manager settings to run the code without any problem ?.
On that day, I could see the LED GPIO1 blinking, not sure what I have done.
What could be the problem ?

Attached arduino board manager settings.
Also tried to use multiple ESP8266 flasher tools, nothing worked out. (xtcom etc.,)

[image]

I could flash the code all the time, but flashed code is not running once I set boot mode as "flash mode"

Can you please help me on this problem.

Regards,
Titus S.

titusece, 2016-10-24:
I am getting attached junk output in serial for every code (LED blink code, WiFi code, Serial print).
[image]

What could be the problem ?

Some times, seen the below exception error.

 _ets Jan  8 2013,rst cause:2, boot mode:(3,5)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x0f
csum 0x0f
~ld

Exception (0):
epc1=0x402066b8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000                                                                                                                                                             000

ctx: sys
sp: 3ffff6e0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffff880:  40101ae9 40101ae6 309cbfff f6ac5105
3ffff890:  400005e1 f76d6d00 d514c9ae ab5aa18c
3ffff8a0:  40216c7f 00000030 00000020 4d31de17
3ffff8b0:  00000000 60000a04 00000001 00000015
3ffff8c0:  00000000 00000052 ffffffe8 00000038
3ffff8d0:  0100157c 00000000 00000000 00000006
3ffff8e0:  00000000 00000000 0000006f 0000001c
3ffff8f0:  00000000 00000000 00000001 40006545
3ffff900:  00000000 00000000 00000000 40006545
3ffff910:  00000000 3ffff9b0 3ffe9518 2e2c2927
3ffff920:  400069d9 00000000 00000000 00000000
3ffff930:  40218259 00000138 00000172 40218145
3ffff940:  00000000 00000000 00000000 00000000
3ffff950:  fffffff0 00000014 40216c63 00000006
3ffff960:  00000000 3ffe955a 00000001 00000000
3ffff970:  3ffff9b0 3ffe955a 00000001 0000001f
3ffff980:  00000040 00000010 00000001 00000033_

I am really very new to this ESP-01 module.....don't how to solve this problem. :(

WereCatf, 2016-10-24:
Using a bare module like the ESP-01 is a hassle, even for a bit more experienced user. Could I suggest that you spend a couple of euro and buy a Nodemcu-board on eBay or DX.com or similar? Like e.g. http://www.ebay.com/itm/NodeMcu-Lua-WIFI-Internet-Things-development-board-based-ESP8266-CP2102-module-M-/152146554889 is a well-tested board and it would be much, much easier for you to get started with and focus on learning the programming side of things. You don't even need a USB-serial adapter with them.

You can always use the ESP-01 later on, when you've gotten some experience with programming on the Nodemcu.

titusece, 2016-10-25:
You are 100% right.
That was also I planned to do.
I also bought ESP-12E (bare-metal board) recently, that I'm not able to flash the code and not even bring up anything.
I have referred lot of documents and links, GPIO15 & GPIO0 should be grounded for flash mode.
I will go with your suggestions.

Thanks million for your support.

Regards,
Titus S.

titusece, 2016-10-25:
Can you please suggest any link to work with Nodemcu board. (what to do for setting boot mode and flash mode etc.,)
I don't want to use the "LUA" firmware, I can flash the LED and WiFi code what we discussed here into Nodemcu board right ?

WereCatf, 2016-10-25:
You don't have to mess with any of that, no need to pull GPIO0 to ground or anything manually. Just connect a USB-cable from your PC to the Nodemcu, select Nodemcu as your board in the Arduino IDE and make sure the COM-port is right, and it does all of that automatically for you when you flash a sketch. That's why I recommended it to you, it's pretty much the best board for a beginner.

titusece, 2016-10-25:
Thanks, I will buy it and try today.
Thank you so much.

BTW, last night, I'm able to see the "Serial.println" code is executing correcly, but not all the time.

void setup() {
Serial.begin(115200);
delay(10);
}
void loop() {
Serial.println("Hello world");

}
Don't know what is the problem, as you said I will go with Nodemcu board.
Thanks again.

titusece, 2016-10-25:
Hi  WereCatf,
I bought the NodeMCU board and all the code is working now (WiFi, LED blinking, serial etc.,).
Thanks for your help.

What is the difference in these board, I see just FTDI part and flash & boot handling in HW.
Right ? is there anything ?

closed 2016-10-25 by titusece

WereCatf, 2016-10-25:
The Nodemcu has 4MB flash, as opposed to the 1MB in your ESP-01. It also has proper power-circuitry on board, including capacitors -- some of your problems with the ESP-01 are most likely because of poor power-supply and no input-capacitors. The ESP8266 is quite sensitive to power-fluctuations and it really wants voltage as close to 3.3V as possible and stable, and that's why good power-circuitry and input-capacitors are important.

Aside from that, no, there isn't much else that is different.

titusece, 2016-10-25:
Thanks for your great answer. :)

rolpal, 2017-10-15:
An old thread/question but the answer to the problem with "black" ESP8266-01, also known as ESP8266-01S is hard to find so I'll add it here to. The 1MB version needs the Flash Mode to be set to "DOUT" in Arduino IDE.

jojje123, 2018-01-23:
do you get it to work ?
I have esp-01s flashed with blynk.
I can toggle onboard gpio01 with blynk when it is powered by the programmer but when I put it in esp-01s relay module it doesnt even boot and connect to wifi
https://s3-ap-southeast-1.amazonaws.com/a2.datacaciques.com/00/MTU5MzA/17/11/03/3y736zt82v08sza8/36499b2638adae95.jpg

rolpal, 2018-01-29:
I have no experience of that relay module.
Check the traditional pin pullups etc. that may stop it to boot from flash.

aledrus, 2018-04-05:
Hi there.

I am not sure if this is what you're asking, but I have just found out that
On the ESP-01; the built-in LED is connected to the GPIO1. 
On the ESP-01S (1mb): the built-in LED is connected to GPIO2.

The serial.prints won't show on the Serial Monitor on the ESP-01 if you've set the pinMode (1, OUTPUT). 
Cheers

Azrudi

loadedlouie, 2018-06-12:
@jojje123 i bought the same board that you. and a lc-technologies. 
I managed to make some code work on a lc-technologies work, but not on this relay boards. 
If you figured this out please share.. im kind of stuck..

electronicsguy, 2018-06-24:
I don't know if this will help anyone, but I am able to blink the onboard blue LED on the ESP-01S (black colored) board, with pin = 1. Sketch attached.
````
void setup() {

  // LED_BUILTIN = 2 doesn't work for ESP-01S. Needs to be "1"
  pinMode(1, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  /*Serial.begin(115200); 
  Serial.print("LED_BUILTIN = ");
  Serial.print(LED_BUILTIN, DEC);
  */
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(1, HIGH);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(500);                      // Wait for a second
  digitalWrite(1, LOW);  // Turn the LED off by making the voltage HIGH
  delay(500);
  digitalWrite(1, HIGH);
  delay(500);                      // Wait for a second
  digitalWrite(1, LOW);  // Turn the LED off by making the voltage HIGH
  delay(500);
  digitalWrite(1, HIGH);
  
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}
````

Also, the "high" and "low" seem to be reversed.

e-xplore, 2018-07-10:
Can anyone provide the code to blink, on, off the LED from a web browser?

aspofficial94, 2018-07-31:
@electronicsguy , can you please please share me your the connection setup or ESP8266-01S with arduino nano. I'm able to upload the sketch properly(atleast that's what the Arduino IDE says), but the led isn't blinking with both GPIO 1 or 2. Please do reply.

Anyone else too, it would be of great help, if you can pass on the pin connections between ESP8266-01S and nano.

Thanks a lot in advance.

electronicsguy, 2018-08-02:
@aledrus & @aspofficial94 There's nothing special about the connections. Some help here: [http://henrysbench.capnfatz.com/henrys-bench/arduino-projects-tips-and-more/esp8266-esp-01-pin-outs-and-schematics/](http://henrysbench.capnfatz.com/henrys-bench/arduino-projects-tips-and-more/esp8266-esp-01-pin-outs-and-schematics/) & [https://www.aliexpress.com/item/5PCS-Mini-ENC28J60-Webserver-module-Ethernet-Shield-board-for-Arduino-Nano-v3-0/1865921970.html](https://www.aliexpress.com/item/5PCS-Mini-ENC28J60-Webserver-module-Ethernet-Shield-board-for-Arduino-Nano-v3-0/1865921970.html).

EvsanDlg, 2019-01-24:
> An old thread/question but the answer to the problem with "black" ESP8266-01, also known as ESP8266-01S is hard to find so I'll add it here to. The 1MB version needs the Flash Mode to be set to "DOUT" in Arduino IDE.
Guys, this is the most important answer for this old thread. Thanks to @rolpal 
And the second important fact for those like me who just started with esp-01 is that built-in LED relates to TXD output which is GPIO1 at the same time. As result, LED_BUILTIN should be =1 for blinking. And, moreover, that means that esp will push HIGH and LOW signals to serial port (because you send HIGH and LOW values to GPIO1 now). That can cause problems with serial print.