esp8266/Arduino #2635: Precise timing for RTC during deepsleep?

nouser2013, 2016-10-24:
dear all, I was wondering if we can have precise and accurate time periods over deep sleep cycles.

I have a sensor node which should measure every 5 minutes. Depending on temperature, two nodes drift 10 to 20 seconds apart each cycle.

I'd like this, so that the sensor node can measure and log to internal file with timestamp, and every 24h get NTP time again and transmit the logfile. Currently, I fire up the WiFi every measurement which of course wastes a lot of power.

Node MCU has this apparently:
https://github.com/nodemcu/nodemcu-firmware/blob/master/docs/en/modules/rtctime.md

WereCatf, 2016-10-24:
The ESP8266 just doesn't have a very precise RTC, if you need precision it'd be best to use an external RTC-module.

nouser2013, 2016-10-24:
Will you read the documentation I pointed out before you make that statement?

WereCatf, 2016-10-24:
I did, and like the documentation says, it is still basically guessing at how much time was spent in sleep. To answer the question, the Arduino-core for the ESP8266 doesn't provide anything like that, but you can always port the code over, if you feel like it.

devyte, 2017-10-11:
The code in the Nodemcu repo is not super precise, but it's _much_ better than nothing. I think porting that code here makes sense.
Leaving this open as a valid feature request, although it may take a long while to get to it.

nbolton, 2022-12-02:
This might be useful: https://arduino.stackexchange.com/a/91504/78174