esp8266/Arduino #2636: SoftAP wont recieve writes from a Client

Makuna, 2016-10-25:
### Basic Infos
Given two Esp8266, where one is configured as a SoftAP (sketch below) and another that tries to connect to it as a client (sketch below), the client will connect to the AP but data is never sent or received.
Further, it seems the softAP settings are not being saved as the channel number isn't updated.

#### Hardware
Hardware:			ESP-12 (tried several NodeMcu and WeMos D1 didn't matter)
Core Version:      	The Arduino Board release 2.3.0 (what ever SDK core that is)

### Description

The writes from the client don't appear on the server, and even seem to take a while (seconds).

### Settings in IDE

Defaults

### Sketch SoftAP
```
#include <ESP8266WiFi.h>
extern "C" {
#include "user_interface.h"
}

const char* ssid = "AP001";
const char* password = "BadDay";

// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(777);
WiFiClient client;

void setup() {

  //ESP.eraseConfig();
  //WiFi.setAutoConnect(false);
  //WiFi.disconnect(true);
  
  WiFi.softAPdisconnect(false);
  
  delay(1000);
  
  Serial.begin(115200);
  while (!Serial); // wait for serial attach
  Serial.println();

  // scan networks to pick a clear channel
  //
  int8_t countNetworks = WiFi.scanNetworks();
  uint16_t countUsingChannel[13];

  for (int8_t indexNetwork = 0; indexNetwork < countNetworks; ++indexNetwork)
  {
    int32_t rssiScan = WiFi.RSSI(indexNetwork);
    int8_t channelScan = WiFi.channel(indexNetwork);
    uint8_t rankScan = (rssiScan + 100) / 5;
    
    countUsingChannel[channelScan - 1] += rankScan;
    
    Serial.print(WiFi.SSID(indexNetwork));
    Serial.print(" (");
    Serial.print(channelScan);
    Serial.print(") ");
    Serial.print(rssiScan);
    Serial.println();
  }
  Serial.println();

  // find an open channel (skip channel 1 completely, too many things default to it)
  //
  uint8_t channel = 2;
  uint16_t channelRank = countUsingChannel[channel - 1];
  for (uint8_t indexChannel = 1; indexChannel < (sizeof(countUsingChannel) / sizeof(countUsingChannel[0])); ++indexChannel)
  {
    uint16_t testRank = countUsingChannel[indexChannel];
    if (channelRank > testRank)
    {
      channelRank = testRank;
      channel = indexChannel + 1; 
    }
  }
  
  delay(500);
  
  WiFi.setPhyMode(WIFI_PHY_MODE_11G);
  WiFi.setOutputPower(15); // low power
  WiFi.mode(WIFI_AP);
  
  // Expose a WiFi network
  WiFi.softAP(ssid, password, channel);

  delay(1000);
  
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());
  
  // Start the server
  server.begin();
  Serial.println("Server started");

  // wait for THE client

  WiFi.printDiag(Serial);
  
  do {
    delay(100);
    client = server.available();
    Serial.print(".");
  } while (!client);

  Serial.println();
  Serial.println("client connected");
}

void loop() {
  
  // Wait until the client sends some data
  
  if (0 == client.available()){
    return;
  }

  Serial.print("D ");
  
  // Read the request
  uint8_t req[2];

  client.read(req, 2);
  
  Serial.print(req[0]);
  Serial.print(" ");
  Serial.println(req[1]);
}

```

### Sketch Client
```
#include <ESP8266WiFi.h>
 
WiFiClient client;
const char* ssid     = "AP001";
const char* password = "BadDay";
uint32_t lastTime;

void setup() {
  //WiFi.setAutoConnect(false);
  WiFi.disconnect(true);
  
    Serial.begin(115200);
    while (!Serial); // wait for serial attach
    WiFi.setOutputPower(10);
    
    WiFi.begin(ssid, password);
    wl_status_t stat = WiFi.status();
    while (stat != WL_CONNECTED)
    {
      delay(500);
      if (WiFi.status() == stat)
        Serial.print(".");
      else
      {
        stat = WiFi.status();   
        Serial.println(""); 
        Serial.println(stat);        
      }
    }
  
    Serial.println();
    Serial.println("WiFi connected");
    Serial.println("IP address:");
    Serial.println(WiFi.localIP());
    IPAddress ipGateway = WiFi.gatewayIP();
    Serial.println("Gateway IP address:");
    Serial.println(ipGateway);
    while (!client.connect(ipGateway, 777))
    {
      Serial.println("connection failed");
      delay(500);
    }
    Serial.println("Client connected");
    Serial.println("Remote IP address:");
    Serial.println(client.remoteIP());
    lastTime = millis();
}
void loop() {
  uint32_t time = millis();
  if (time - lastTime > 10000)
  {
    lastTime = time;
    uint8_t status = client.connected();
    Serial.print(status);
    
    client.write(1);
    client.write(4);
    client.flush();
    Serial.println(" D 1 4");
  }
}

```

### Debug Messages
(network names x out for privacy)
```
xxxxxxxxxxx (1) -81
xxxxxxxx (1) -40
xxx (6) -92
xxxxxxxxxx (9) -92
xxxxxx (11) -87
xxxxxxxxxx (11) -89
xxxxxx (1) -89

AP IP address: 192.168.4.1
SSID: 
Server started
Mode: AP
PHY mode: G
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
```

Makuna, 2016-10-27:
Issues found.
1) Must force the AP into a AP mode by calling `WiFi.mode(WIFI_AP);  `
2) AP passwords if provided must be 8 chars or longer or the call to `WiFi.softAP()` will fail. So either empty string for no password or something that is at least 8 chars.

closed 2016-10-27 by Makuna

Juppit, 2016-10-27:
Finally, point two should be included in [esp8266wifi/readme.md](https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/readme.md).