esp8266/Arduino #2638: Very Slow big file SPIFFS read out serial port ... intermitent

Mark-Tec-Wizard, 2016-10-25:
I am bringing down a very large 1mb hex file from my ftp server.  It is being used to IAP an attached embedded system.  When I read the file out the serial port the most times it streams the file out without delay between characters.  Intermittently the same file that is brought down to the file system will abruptly start to send the characters out with big delays.  Most times deleting the file and getting it from the FTP server will fix the problem.  To be clear, a file that works always works,  a file that is slow always is slow.

A slow file is fast at the beginning but abruptly goes slow at different sizes.  Not sure if reformatting helps.

Any sugestions

Module:  			Huzza
Flash Size: 		4MB
CPU Frequency:	80Mhz

WereCatf, 2016-10-26:
The SPIFFS used by this Arduino-core is a tad outdated, I have newer SPIFFS in my version of the ESP8266 Arduino-core at https://github.com/WereCatf/Arduino that you could try and see if it fixes your issue or not. I really don't know if it helps or not, but there was another guy here some time ago who had issues with SPIFFS and using my repo fixed it for him.

Mark-Tec-Wizard, 2016-10-26:
Thank you for the response.  I am unsure of how to update the version.
When I add the http://arduino.esp8266.com/stable/package_esp8266com_index.json to the additional board manager url
I cant seem to find it on the  Board Manager dialog box.  I am currently using a HUZZA board.
I read the readme file in the newer version has a fix for large file reads so I think it should fix the problem.  How can I update the library to get the new spiffs version to compile.

Thanks
Mark

WereCatf, 2016-10-26:
Either use a git-client, or just copy all the files in cores/esp8266/spiffs/ on my repo over any existing files in  cores/esp8266/spiffs on your installation.

Do report back if it fixes your issue.

Mark-Tec-Wizard, 2016-10-26:
Thanks very much.
 I will let you know if it worked.  It is intermittent but happens quite often so I should know if it solved the issues by the end of today.

Mark-Tec-Wizard, 2016-11-04:
Replaced spiffs files and the same problem persists.  I plan to store the file in 100k chunks and when the embedded system asks for the file send it out the serial port as one file.   I would like to solve the underlying issue though.   The problem occurs about 75 percent of the attempts of down loading files 750k and 1.5m    Again I am using a new adafruit huzzah board.

comino, 2016-11-22:
Same observation on my side as Im downloading a 450k file. It slows down over time. 
But I wouldn't blame SPIFFS - it looks like a lwip issue, since the sockets seems to parse fewer and fewer data over time. 

https://github.com/esp8266/Arduino/issues/2700
https://github.com/esp8266/Arduino/issues/2689

might be correlated

Mark-Tec-Wizard, 2016-11-22:
Comino - I am making an ftp connection and then saving the file to a spiffs file.  I disconnect then read the file.  It reads the first part fast then slows.  I wrote a routine that breaks the spiffs file down into smaller files and it does the same thing on the first file.  I am working on a spiffs test that will write big files and see if I can repeat the problem not saving the data from ftp connection.  Ill let you know the results.

themindfactory, 2016-11-22:
I brought this to attention a few weeks back also... read my post http://www.esp8266.com/u/oddspiffs

RichardS

Mark-Tec-Wizard, 2016-11-30:
Here is some test code that we wrote to show the problem in its simplest form.
The program creates a big sipff file then reads it back.  About 50% of the time it will create a file that reads back slow.
*********** ABSTRACT ***********************************
This program creates a 26000 line file where each group of ten lines is formatted as shown below:
Line00001:11111111111111111111111111111111
Line00002:22222222222222222222222222222222
Line00003:33333333333333333333333333333333
Line00004:44444444444444444444444444444444
Line00005:55555555555555555555555555555555
Line00006:66666666666666666666666666666666
Line00007:77777777777777777777777777777777
Line00008:88888888888888888888888888888888
Line00009:99999999999999999999999999999999
Line00010:00000000000000000000000000000000
Line00011:11111111111111111111111111111111
...
Line26000:00000000000000000000000000000000

After creating the file, the program reads it back from the SPIFFS and sends it out the serial port.

In many attempts but not all, at some point in the file, the serial output slows down to about 1 line every 2 seconds.  
The point in the file where it slows down is repeatable for a given file but varies between files.

The first time running the test, the source is compiled with the SPIFFS formatting uncommented and loaded.
Once the SPIFFS has been formatted and one instance of the file has been created, 
the program is recompiled without the SPIFFS formatting and reloaded.

After the file is deleted and recreated, it is read and output to the serial port at 115200 baud.  
It outputs many lines per second until it completes or slows down to about 1 line every 2 seconds.  
If it slows down to 1 line every 2 seonds, manually reset the HUZZAH board and try again.

We have run these tests and observed this behavior on two different HUZZAH boards.
****************************************************************************************************/
#include "FS.h"

void setup() {
  char strTemp[64];
  int i=0;
  
  Serial.begin(115200);

  // always use this to "mount" the filesystem
  bool result = SPIFFS.begin();
  Serial.println("SPIFFS opened: " + result);

/*  
  Serial.println ("* Beginning to format...");
  Serial.print(" ...  will take up to 1 minute to format");
  Serial.println();
  SPIFFS.format();
  Serial.println("  Format complete");
*/

  if (SPIFFS.remove("/f.txt")) 
  {    
        Serial.print("Deleted file  ");
        Serial.println("/f.txt");
  }else
  {    
        Serial.print("/f.txt");
        Serial.println("   File not found ");
  }

  Serial.println("File doesn't exist yet. Creating it");

  // open the file in write mode
  File f = SPIFFS.open("/f.txt", "w");
  if (!f) {
    Serial.println("file creation failed");
  }
  else
  { //create the file
   sprintf(strTemp,"Line%05d:00000000000000000000000000000000\r",i);
   for(i=0;i<26000;i++)
   {
      char c=strTemp[10];
      c++;
      if(c>'9') c='0';
      sprintf(strTemp,"Line%05d:00000000000000000000000000000000\r",i+1);
      for (int j=0;j<32;j++)
      {
        strTemp[10+j]=c;
      }
       f.println(strTemp);
    }
    f.close();
    // read back the file and send it out the serial port    
    f = SPIFFS.open("/f.txt", "r");
    if (f) 
    {
      // we could open the file
      while(f.available()) {
        //Lets read line by line from the file
        String line = f.readStringUntil('\n');
        Serial.println(line);
      }
  
    }
    f.close();
  }
}

void loop() {
  // nothing to do for now, this is just a simple test

}

devyte, 2017-10-11:
@Mark-Tec-Wizard is this issue still valid with latest git? Please test and report back here.

devyte, 2018-07-12:
Closing due to lack of feedback and age.

closed 2018-07-12 by devyte