esp8266/Arduino #2641: Is ESP32 supported on this project ? On future

asetyde, 2016-10-28:
New big boss is coming , at now first product it is coming ..
[image]

It will support ESP 32 ?

me-no-dev, 2016-10-28:
ESP32 Arduino [HERE](https://github.com/espressif/arduino-esp32)

jrobeson, 2016-10-30:
@me-no-dev : seems like they could share some amount of code, especially in the case of the http libraries that aren't hardware specific.  How should that be done?

igrr, 2016-10-30:
@jrobeson we are planning to align our core network APIs in the ESP32 core with official Arduino libraries, to improve platform compatibility. So while there certainly is a place for code reuse, it may not be as straightforward as just renaming wifi library include files.

jrobeson, 2016-10-30:
So in the near term do you plan on having full copies of say https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp ?

jrobeson, 2016-10-30:
Alternatively, could just use this thing (were it to have TLS). https://github.com/arduino-libraries/ArduinoHttpClient

asetyde, 2016-11-02:
Thx all to reply , i think is imprint not lose big work on esp8266 and give it to esp32 , esp8266 continues to be best chip for cheap and fast IoT , we need more for some specific applications but for 80% is good.

Humancell, 2016-12-06:
We are fully committed to using the ESP8266 at this point ... but want to keep aware of the possible future support for the ESP32, as a possible future migration path.

Can someone make a brief statement about the two projects that we've been tracking?

We're using this one today (https://github.com/esp8266/Arduino) and love it!  We've been tracking the https://github.com/espressif/arduino-esp32 project, but it still appears to be much further behind, and not completely compatible.

When I saw the comment above: "we are planning to align our core network APIs in the ESP32 core with official Arduino libraries, to improve platform compatibility. So while there certainly is a place for code reuse, it may not be as straightforward as just renaming wifi library include files."

I'm curious exactly what this means, and the extent of compatibility in the future between the ESP8266 and ESP32 code?

me-no-dev, 2016-12-06:
Code will be pretty much compatible, but there are some major differences in both chips that will bring some differences in the code as well. Both projects are maintained by the same people and while it's true that ESP32 is still behind on many things, do not forget that ESP8266 has been around for a while now and have gained much momentum, while the ESP32 is still under heavy development in the core, that all Arduino is built on top of. You probably do not know how long it took to be possible to have Arduino on ESP8266 :)

Humancell, 2016-12-06:
Hello,

First, please do not take my questions or comments to be ungrateful or think that I am underestimating the effort that has been put into both projects!  I've been following the ESP8266 project since early on and am astounded by the progress.  I am also a developer, but not quite as skilled as you, igrr and the others contributing here ... but I do recognize the work and incredible effort put into this so far.  :-)

Thank you for this ... and also, where is the "Donate" button to make a difference?

Being that you and the same people were working on both, I was trying to understand the "plan" or "intentions" for this code and the ESP32 version.  Is anything documented that might set expectations on the implementation for the ESP32 (not necessarily time frames) and how you see the two code bases moving forward?  Will it eventually be one?  Or two, based on Board selection? abstracting APIs to deal with differences?

And your answer could be "no" ... I was mostly just curious as the ESP32 continues to develop momentum, and has a few features that are very nice for future designs.

Thank you again for all of the work!

me-no-dev, 2016-12-06:
Intentions are all things that run on the ESP8266, to also run on the ESP32, and then some :)
API here and there (WiFi mostly) might slightly change in order to align with official Arduino API. I think it's pointless to explain why that is important :) Rest assured though, you will not lose any functionality and if possible we will have deprecation notices a few versions before totally ditching "old" API.
And then there is the case where we have so much new hardware that Arduino has no notion or API for... Example: there are 2 types of hardware PWM + SigmaDelta + DAC. Arduino has only analogWrite().

closed 2017-07-12 by asetyde