esp8266/Arduino #2643: ESP8266 reset when include

sonyarouje, 2016-10-28:
ESP8266 reset when adding #include "ESP8266WiFi.h". My code works fine without #include "ESP8266WiFi.h" and I can see hello world printing to my serial monitor.

I use a USB to FTDI module to upload code.

My ESP module is '[ESP8266 Serial ESP 01](http://www.ebay.in/itm/272282635593?aff_source=Sok-Goog)'

Not sure what is wrong here. I tried uploading the sketch via Arduino IDE 1.6.7 and 1.6.12.

Any help will be really appreciated.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512 (64k)
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

Below is my code.

#include "ESP8266WiFi.h"
void setup() {
// put your setup code here, to run once:
Serial.begin(115200);
}

void loop() {
// put your main code here, to run repeatedly:
Serial.println("hello world");
delay(1000);
}
```

### Debug Messages

Error seeing in Serial monitor is

ets Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

WereCatf, 2016-10-28:
It obviously shouldn't crash just from that. What version of the Arduino-core for ESP8266 are you using?

One thing that comes to my mind is that people often have troubles with generic ESP8266-modules due to bad power-supply and/or no capacitor between VCC and GND -- how are you powering it?

sonyarouje, 2016-10-29:
I tried with IDE 1.6.7 and 1.6.12, same issue from both ide.

For power supply I used a external regulated supply, which provides close to 3.28v. So to eradicate any power issues I connected to 4.5v as well (I know it's not safe with 4.5v). Both scenarios same issues.

Yes I do have a 470mf cap at VCC/GND.

WereCatf, 2016-10-29:
Just including a header should not change anything, let alone cause crashes, and I cannot replicate your problem on my devices. I do not know what's wrong with yours.

sonyarouje, 2016-11-01:
I am on holiday for last couple of days, I will do some more experiments and come back to you all.

sonyarouje, 2016-11-07:
I fixed the issue. This is what I did.

I did several attempts to rectify the issue. Nothing really worked. So I decided to burn the AT bootloader back to the module. When I try to bootload with AT, I started getting fatal exception. Then I switched my USB FTDI module, this time I could bootload successfully. Using the same USB Serial I burn the Arduino sketch again and it worked very well. 

I am not sure whether the issue was related to my USB Serial or a corrupted flash memory.

Thanks WereCatf for your suggestions.

devyte, 2017-10-11:
Per previous comment, closing as resolved.

closed 2017-10-11 by devyte