esp8266/Arduino #2644: Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json

edautz, 2016-10-28:
Since half an hour I get the message:

http://arduino.esp8266.com/stable/package_esp8266com_index.json using Arduino IDE 1.6.8. with boardmanager 2.0.0.

However when I try this link in a browser, I an JSON file. 

This morning everything worked OK.

I disabled my PC firewall and virusscanner, but the error stays.

Could there be a syntax error in the http://arduino.esp8266.com/stable/package_esp8266com_index.json 

file?

edautz, 2016-10-28:
After upgrading to arduino IDE 1.6.12, I got the following error:

Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json
Invalid version found: 1.04
Invalid version found: 1.04

edautz, 2016-10-29:
No one has this problem too? I guess a lical problem toen on my setup.

edautz, 2016-10-29:
Was a local problem, on the Arduino preferences dir the file package_esp8266com_index.json was 0 bytes big and someway locked. After deleting this file everythings worked OK.

closed 2016-10-29 by edautz

rizalul, 2017-01-13:
@edautz I get the same problem with you. So, how to solve the problem sir?

edautz, 2017-01-15:
This file is downloaded to your system when using the IDE. 

On my laptop the file was corrupted and 0 bytes in size. After deleting it and restarting the IDE the problem was solved. You can find the file in your Arduino dir tree on your system. 

Verstuurd vanaf mijn iPhone

nickjan, 2017-07-26:
can not donload esp8266 getting json error again and again what to do?

nickjan, 2017-07-26:
While opening toos ->board manager it shows http://arduino.esp8266.com/stable/package_esp8266com_index.json error

edautz, 2017-07-26:
In my case the local json file on my laptop was corrupted. After deleting it the problem was solved. 

See the older comments of this issue to see the file location details.

arthur12320, 2017-11-17:
i`m having the same problem and i tried deleting the preferences file and it had no effect

dontsovcmc, 2018-02-20:
https://github.com/esp8266/Arduino/releases/download/2.3.0/package_esp8266com_index.json =)

Sunbelife, 2018-08-17:
you are good man

dianamichel, 2018-10-20:
desafortunadamente aun sigue sin funcionar y el archivo json local, no se donde esta para eliminarlo como dicen aca.

nisho5, 2018-10-28:
Running Arduino As Administrator fixed the problem for me :)

sandeepguptapra, 2019-06-11:
i have problem in error down loading http://arduino.esp8266.com/stable/package_esp8266com_index.json

Surkhab-singh, 2019-09-06:
This solved my problem:
Just go to windows drive >> Users >> user account name you are using >> AppData>> Local >> Arduino15 and delete files with esp8266 in their name.

Note: AppData is normally hidden by default
If this doesn't work then delete all the files in Arduino15 folder

hack-tramp, 2022-08-10:
None of the above worked for me. What I did was I deleted the files in AppData\Local\Arduino15\staging\packages and downloaded the required files manually, into that directory. This is what I have: 

esp8266-3.0.2.zip
i686-w64-mingw32.mklittlefs-943d2f7.210717.zip
i686-w64-mingw32.mkspiffs-7fefeac.210717.zip
i686-w64-mingw32.xtensa-lx106-elf-1757bed.210717.zip
python3-3.7.2.post1-embed-win32v2a.zip

Then I open the IDE and it just installs, it doesn't need to download anymore