esp8266/Arduino #2645: Short Pulse in Input GPIO when reset Occurs

cyguevara, 2016-10-28:
Hardware:			ESP-12F
Core Version:      	2.3.0

I use a GPIO send a signal to the ESP8266, I configure the port as Input (No Pull_UP), and after sending some data to a server it goes to deep sleep, and the system wakes up every 30 seconds automatically, and repeat the process.

Problem description
Every time the system starts again or wakes up the input GPIO generates a short positive pulse. How can I avoid this pulse?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz?
Upload Using:		 SERIAL
Reset Method:

jrobeson, 2016-10-30:
you didn't mention which pin you used.

RudyFiero, 2016-10-31:
Most of the pins are started with pull up resistors, or have some activity. The only ones that don't are D4 and D5. All of this happens in the bootloader code (Expressif)  and has nothing to do with the ESP-Arduino code. It sucks but it is just something we have to live with.

devyte, 2017-10-11:
@cyguevara this is normal, and it can't be avoided because it's in the Espressif SDK and not in the Arduino core.
Closing.

closed 2017-10-11 by devyte