esp8266/Arduino #2647: Enhancement: More "simple" web protocols?

Matthew-Bradley, 2016-10-28:
### Basic Infos
Enhancement: Support for more basic web protocols out of the box

### Description
Hi,

I was wondering if it would be possible (manpower permitting) to implement a few useful protocols, considering things like TLS and https work. Some ideas:
-IRC
-Email sync (sending and receiving)
-FTP (if not implemented already)
-any of a number over-the-network messaging protocols

I'm sure other people can come up with a ton more "simple" but tremendously useful protocols. Or, if anyone knows of existing implementations which can be adapted to the ESP, could they be added to list of known working libraries? 

Having such a capable device opens the doors to being compatible with existing standards, instead of hacking together something for a one-off use (like hardcoded GETs), but I don't know how many well-implemented protocol libraries have been written in the dark days of underpowered unos with ethernet shields...

P.S. Whoever wrote the existing library for making HTTP POSTs and GETs, this is amazing work and I hope that other protocols can receive equal treatment in the future!

jrobeson, 2016-10-30:
I'd hope those would at least attempt to be generically for arduino and not tied to the esp8266/esp32. So it'd be better if they were external libs if possible.

Matthew-Bradley, 2016-10-30:
I don't think it would be possible to even attempt that, since there's a huge difference in processing power between an ESP8266 and other arduino-compatibles.

I don't know of a single other arduino compatible capable of running https, or even TLS. Without those for example, STMP, IMAP, and POP3 won't work since all modern email servers require a secure connection. Now that servers are willing to listen to an ESP8266, we don't have to mess around with the weird, hacky solutions often employed to get other boards on the web, like trying to proxy connections through a raspberry pi or use a third party service to gain the appearance of https. It seems to be hugely counterproductive to even attempt to port any of this to older boards that can't handle the required protocols.

jrobeson, 2016-10-30:
Sorry, i should have been more specific. I mean arduino the platform (and apis), not specific hardware like the Uno.  There are ports to various arm boards that have near or more the specs of the esp8266, so these libraries would work equally well there, and then there's also the esp32.

Matthew-Bradley, 2016-10-31:
A good question (which this would be dependent on) would be how standardized, and how adherent to existing arduino "standards," the ESP8266's WiFi libraries are or will eventually be, especially as more functionality such as TLS gets brought in. As far as I can tell, as lot of it isn't very well documented (yet) so it's difficult to give a good answer without combing through the relevant code.

Certain aspects certainly seem weird in comparison with how arduino does it (which isn't even consistent between the Ethernet shield and WiFi shield), but if it's well organized and makes sense once documented, then I think it's possible that what gets put in place on the ESP8266 might become the template for when other more power full ARM based chips (and other arches) eventually implement the same features. (which means that some real thought ought to be put into how basic to medium-level functionality is exposed and called)

Matthew-Bradley, 2016-10-31:
One example is having consistency between how different kinds of WiFi networks are connected to, using overloaded functions that have similar signatures. Another would be in having parity between how server and client sides of a given protocol are implemented. Yet another would be in having functionality divided into distinct, preferably replaceable, layers. That way, you could cleanly swap unneeded HTTP/HTTPS functionality for STMP, IMAP, and POP3 if your application only interacts with email servers and never fetches webpages. More extreme, you might eventually want to replace the packet driving(?) code some day if an ESP appears with a (not hacked in) Ethernet interface or one is added using an external chip over SPI (already possible with the Pi, and theoretically possible with the Ethernet Shield, so it's not completely out of  the question), and you suddenly have only Ethernet instead of WiFi, or even worse, both Ethernet and Wifi, and might want to pick which to use with the same libraries and code. 

This might seem (and probably is) a long way off (why have anything more than a single WiFi interface?), but a bit of structural planning now will probably pay dividends later after a bazillion libraries have been written by third parties, which might otherwise be rendered unusable by circumstance.

One of the major shortcomings of arduino has been how completely boned you are if two libraries both use the same resources, like the same timer, or one monopolizes the SPI or a serial port. That might not be easy to fix while maintaining language compatibility with arduino, but it is something that could probably be avoided in future with well thought-out conventions. Some of the most forward-thinking libraries I've ever seen have been ones that simply accepted an SPI or Serial object, and made the required standardized calls to it. That way you could hand it say, Serial2 on an arduino mega, or could just pass Serial on an Uno.

Humancell, 2016-10-31:
I'm curious ... did you not find a few examples of these higher-level protocols when Googling?  I did some quick searches and found a few ... but haven't tested them ... and they seem to just build on the client sockets?

IRC - https://github.com/Fredi/ArduinoIRC
SMTP - http://playground.arduino.cc/Code/WiFiEmailClient
POP3 - http://playground.arduino.cc/Code/EmailPOP3
POP3 - http://playground.arduino.cc/Code/POP3Email
FTP - http://playground.arduino.cc/Code/FTP

All of these are pretty trivial protocols over a TCP socket ... is there more that you are looking for?  Or just looking for a more comprehensive and better integrated solution?

BTW - we are actually now building products with the ESP8266 that are ONLY using the WIfi for configuration ... some of them never connect to a Wifi network.  In fact, we're completing a solution now that is a Ethernet device that you configure via Wifi after pressing and holding a button.  It allows a user with physical access a very simply way to configure the device from their mobile device.

WereCatf, 2016-10-31:
@Humancell Those SMTP- and POP3-examples don't support SSL/TLS, which seems to what the OP is after. There are very few SMTP- and POP3-servers anymore these days around that support access without SSL/TLS. Oh, and I think the OP would like them to be turned into proper libraries, which isn't a bad idea IMHO.

At least the FTP-example doesn't require SSL/TLS and would be quick and easy to turn into a library in and of itself. The IRC-client seems to already support ESP8266, I took a look at its header, so that's a nice find.

RudyFiero, 2016-10-31:
I have been hoping for some good examples of SSL/TLS. But can the ESP8266 have the resources? Or will this be the reason to move to the ESP32?

WereCatf, 2016-10-31:
@RudyFiero The ESP8266 Arduino-core does ship with axTLS and there is an example on how to use HTTPS (which works just fine, so yes, it does have the necessary resources.) There are no other examples, as far as I know, but you can always dig in to the source-code to figure out how to use it.

RudyFiero, 2016-10-31:
I should look before I ask, but I'm at work and busy. What I was unsure of is if the ESP8266 can be a HTTPS server. I thought it could as a client but I have not looked at the example yet so I don't know what is there.

Matthew-Bradley, 2016-11-01:
I have a few example of some simple stuff, for talking to google scripts over https, in a repo.

devyte, 2017-10-11:
Very interesting discussion, and I'll probably have to look into some of these protocols at some point, but this is the wrong place to discuss. Closing per #3655 .

closed 2017-10-11 by devyte