esp8266/Arduino #2653: MB85RC16 Fram memory

porcao, 2016-11-01:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	?2.1.0-rc2?

### Description

I tried to record two numbers in two memory locations. But every recording is made and read from the same address.
If we try to read one byte, for example, a completely different place it always returns me first byte that is written at the beginning of the memory.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

#include <Wire.h>
int fram = 0x50;
uint16_t writeaddress = 0x20;
uint16_t testaddress = 0x30;
uint16_t testaddress2 = 0x40;

void setup() {
  Wire.begin();

  Serial.begin(115200);
  while (!Serial) ;
  Serial.println("Starting...");

  Wire.beginTransmission(fram);
  writeAddr(writeaddress);
  if( Wire.endTransmission() > 0) Serial.print("Erro da Fram");
  Wire.requestFrom(fram, 1);
  Serial.print("Leitura 1:"); Serial.println(Wire.read(), DEC);

  Wire.beginTransmission(fram);
  writeAddr(writeaddress);
  Wire.write(77);
  Wire.endTransmission();

  Wire.beginTransmission(fram);
  writeAddr(writeaddress);
  Wire.endTransmission();
  Wire.requestFrom(fram, 1);
  Serial.print("Leitura 2:"); Serial.println(Wire.read(), DEC);

  Wire.beginTransmission(fram);
  writeAddr(testaddress);
  Wire.write(21);
  Wire.endTransmission();

  Wire.beginTransmission(fram);
  writeAddr(writeaddress);
  Wire.endTransmission();
  Wire.requestFrom(fram, 1);
  Serial.print("Leitura 3:"); Serial.println(Wire.read(), DEC);

}

void writeAddr(uint16_t addr)
{
  uint8_t highADDR = (uint8_t)((addr & 0xFF00) >> 8);
  uint8_t lowADDR = (uint8_t)(addr & 0x00FF);
  Wire.write(highADDR);
  Wire.write(lowADDR);
}

void loop() {
  // nothing to do
}

Result:
Starting...
Leitura 1:21
Leitura 2:77
Leitura 3:21  

The correct is "Leitura 3:77 "

edgiru, 2017-05-08:
Show the wiring diagram

devyte, 2017-10-11:
@porcao interesting, but this is not the right place to discuss your particular project.
Closing per #3655 .

closed 2017-10-11 by devyte