esp8266/Arduino #2655: Unable to use SPIFFS

sxkod, 2016-11-01:
### Basic Infos
Hi there

I have been trying to use SPIFFS. I will paste the code below. I am somehow unable to format the flash and not able to mount/create/write/read. Any guidance on finding the trouble will be great. I get a fail at SPIFFS.format(). I tried doing format() only in one cycle in setup() and then comment it out among other things. 

Thanks

 
  

#### Hardware
Hardware:	        ESP-12e on nodemcu 1.0
Core Version:      	2.3.0-rc2 
Arduino ide 1.6.5-r5

### Description

Formatting SPIFFS fails.

### Settings in IDE

Module:  			Generic ESP8266 Module 12e
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio?
Flash Frequency:	80Mhz?
Upload Using:		USB
Reset Method:		 nodemcu

### Sketch

```
int myfs_mount(){
  int formatted=0;
  if (!SPIFFS.begin()){
    formatted=SPIFFS.format();
    if (!formatted){
      return -1;}
    else {
      SPIFFS.begin();
      return 1;}
    }
  else{return 1;}
}

String myfs_read(const char* fname){
  int mntd=myfs_mount();
  Serial.print("mntd ");
  Serial.print(mntd);
  if (!mntd) {return "";}
  char istr[64];
  
  File f = SPIFFS.open(fname, "r");
  Serial.print("open ");
  Serial.print(mntd);

  if (!f) {return ""; }
  Serial.print("opened ");
  Serial.print(mntd);

  f.seek(0,SeekSet);
  int rstat=f.readBytes(istr,63);
  if (rstat>0){return (String)(istr[0]);}
  f.close();
}

int myfs_write(const char *fname, const char* v){
    int mntd=myfs_mount();
    if (mntd==-1) {return NULL;}
    File f = SPIFFS.open(fname, "w+");
    if (!f) {return NULL; }
    f.print(v);
    f.close();
    return 1;
}

  myfs_write("/test.txt","abcd");
  Serial.println("reading file");
  Serial.println(myfs_read("/test.txt") );
```

sxkod, 2016-11-03:
Any takers?
Thanks all

mailforkeith, 2016-11-04:
I have problem with SPIFFS create failure. I have tried to get OTA and SPIFFS data upload from ESP8266 FSBrowser example woking on OSX 10.7.5 under Arduino IDE 1.6.7 and 1.6.8, using ESP8266 core boards pkg 2.2.0 and 2.3.0 with the same errors. 

I am not very knowledgeable about Linux, but I have attempted to explore some command line uses in Terminal as well.

I have tested this with with ArpilBrother CP2102 ESP-01 dedicated programmer and also a ESP-12E Witty Cloud board with Micro USB adaptor plate, but both setups give the same error. I have run flash config check sketch to match my selected flash sizes to the hardware each time. 

I can upload sketches successfully to both, and the FSBrowser sketch connects to my router, gives IP address, and when I go there with Safari browser, it says no files because I can't load the index.html from data folder, so part of it is working. My process is that I have the serial console closed, board connected, settings selected (board type, flash size, port), the flash button(GPIO-0 --> GND) depressed on power up, and then click "Tools > ESP8266 Sketch Data Upload" button in IDE (I have also tried releasing the button at different times as well, but that is not needed for flashing the sketch to load).

I have Googled, read related threads, tried the modified Platform.txt methods, but still no joy.

I hope someone on this board can help. Thanks in advance.

examples of errors:

Arduino: 1.6.8 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, QIO, 115200, 1M (128K SPIFFS), ck, Disabled, All"

[SPIFFS] data   : /Users/kd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WebServer/examples/FSBrowser/data
[SPIFFS] size   : 128
[SPIFFS] page   : 256
[SPIFFS] block  : 4096

SPIFFS Create Failed!

----

Arduino: 1.6.8 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

[SPIFFS] data   : /Users/kd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WebServer/examples/FSBrowser/data
[SPIFFS] size   : 3052
[SPIFFS] page   : 256
[SPIFFS] block  : 8192

SPIFFS Create Failed!

mailforkeith, 2016-11-04:
In case this helps someone else, I was just lucky this morning to have solved my problem of not being able to create SPIFFS (previous msg, above). 

This morning I found that I had two versions of the "mkspiffs" tool on my hard drive. The default "mkspiffs" tool is 276kb and came as part of the boards manager download, but the other one I got yesterday during my efforts to fix the "SPIFFS Create Failed!" error from GitHub - WereCatf/mkspiffs/
I noticed that WereCatf's file was 355kb, so I replaced the default one with WereCatf's tool, and now the SPIFFS image is created and uploaded without problems!

Thanks to WereCatf… and please can someone look into this and fix the 2.2.0 and 2.3.0 boards manager accordingly.

sxkod, 2016-11-07:
Hi mailforkeith, I am glad it worked for you. So, do I have to use mkspiffs on each and every board then? I was under the impression, the fs can be made by internal code via FS.format(). Perhaps that is where I missed it!

mailforkeith, 2016-11-11:
Hi sxkod, from my limited experience, I believe it can be done either way. I would check the commands from the examples or the Wiki, but I think FS.Format() is to erase the whole SPIFFS so you can start over fresh. 

I was working with an example in the WifiManager library ( https://github.com/tzapu/WiFiManager ) that uses a method within the sketch to create, read, and write, to SPIFFS for small data, which can be used much like EEPROM, but the other method is described in the ESP8266.com Arduino IDE Wiki and uses the "Upload SPIFFS Data" link in the IDE "tools" menu to load files from the "data" folder in the sketch directory. This method is useful when loading larger data or multiple files, such as demonstrated in the ESP8266 FSBrowser example or for operating a web page from SPIFFS hosted web pages. 

Cheers!

sxkod, 2016-11-14:
Hi mailforkeith

Sadly, still no joy. I tried both as a generic esp8266 and modified options in many permutations such as 
40 dio 4/1 , 40 qio 4/1 , 80 dio 4/1 , 80 qio 4/1 , 40/dio 4/3, 80/dio 4/3, 40/qio 4/3, 80/qio 4/3.  I also tried more than one board just in case, with and without external supply of power too.

Code is below. Unfortunately nothing works. The code below generates the binary chop debugging as below.

Debug messages:
Started SPIFFS : 0
SPIFFS formatted : 0
Opening file
File doesn't exist yet. Creating it
File creation failed

+++++++++++++ code below

```
#include "FS.h"

void setup() {
  Serial.begin(115200);
  
  bool result = SPIFFS.begin();
  Serial.print("Started SPIFFS : ");
  Serial.println(result);
 
  result=SPIFFS.format();
  delay(30000);
  Serial.print("SPIFFS formatted : ");
  Serial.println(result);

  // this opens the file "f.txt" in read-mode
  Serial.println("Opening file");
  File f = SPIFFS.open("/f.txt", "r");
  
  if (!f) {
    Serial.println("File doesn't exist yet. Creating it");

    File f = SPIFFS.open("/f.txt", "w");
    if (!f) {
      Serial.println("File creation failed");
    }
    // now write two lines in key/value style with  end-of-line characters
    f.println("ssid=abc");
    f.println("password=123455secret");
  } else {
    // we could open the file
    while(f.available()) {
      //Lets read line by line from the file
      String line = f.readStringUntil('\n');
      Serial.println(line);
    }

  }
  f.close();
}

void loop() {
  // nothing to do for now, this is just a simple test

}
```

mcr-ksh, 2017-05-11:
+1. same issue here.

BobPrust, 2017-05-19:
I want this code to execute
SPIFFS.begin(); 
    FileName = "";
    FileName = MTF;     
    if (!SPIFFS.exists(FileName)) {    
            MakeDDFFile();      //   make DDF 
            MakeTFFiles();      //    make Daily and Monthly Table files 
Serial.println("Made Startup Files on Start-up");
    }

SPIFFS.end();

To get it to work correctly I need to use;
 if (SPIFFS.exists(FileName)) ......??SPIFFS and !SPIFFS inverted??

I format() the ESP12E fine.  Erase all files. Yep sizes() are all 0's
Ask to make files if they don't exist and nope -
Ask to make files if they exist (and they don't)  works fine....

j0hnsmith, 2017-05-30:
I had this problem with Wemos d1 mini and d1 mini pro, flashing with `esptool.py write_flash --flash_size 32m ...` seems to do the trick. 

I _know_ that these boards have 4 & 16mb respectively (not 32mb) but it works for some reason, correct boundary address I suspect.

jandrassy, 2017-06-15:
32 megabits is 4 megabytes

j0hnsmith, 2017-06-16:
> 32 megabits is 4 megabytes

Of course, I assumed it was in megabytes.

```
$ esptool.py write_flash --help
...
--flash_size {detect,4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}, -fs {detect,4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}
                        SPI Flash size in Mbit
```

suculent, 2017-08-11:
I have same issue. Including code that checks realSize vs. ideSize, it both crashes inside SPIFFS.format() and SPIFFS.begin().

```
void setup() {

  Serial.begin(115200);
  Serial.setDebugOutput(true);

  while (!Serial);
  delay(500);

   Serial.print("* TH: SPIFFS real size = ");
   Serial.println(realSize); Serial.flush();

   Serial.print("* TH: SPIFFS IDE size = ");
   Serial.println(ideSize); Serial.flush();

   Serial.println("* TH: Checking SPIFFS:"); Serial.flush();

   bool flashCorrectlyConfigured = realSize.equals(ideSize);
   if(flashCorrectlyConfigured) {
      Serial.println("* TH: Formatting SPIFFS..."); Serial.flush();
      SPIFFS.format();
      Serial.println("* TH: Formatting completed. Calling SPIFFS.begin in 1sec..."); Serial.flush();
      delay(1000);
      fileSystemReady = SPIFFS.begin();
      Serial.println("."); Serial.flush();
   }  else {
     Serial.println("flash incorrectly configured, SPIFFS cannot start, IDE size: " + ideSize + ", real size: " + realSize);
   }
}

with a stacktrace:

Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 6 results
0x4020175a: Print::write(char const*) at ?? line ?
0x4020175a: Print::write(char const*) at ?? line ?
0x4020428b: SPIFFSImpl::begin() at ?? line ?
0x4020329a: delay at ?? line ?
0x402018c9: Print::println(char const*) at ?? line ?
0x402151f3: setup at ?? line ?
```

devyte, 2017-10-11:
@sxkod I am using nodemcu and SPIFFS, and I can't reproduce this issue with latest git. Format works fine, and mkspiffs + upload from Arduino IDE works fine.
Closing as can't reproduce. If you think the problem is still valid, please open a new issue, and add updated details.

About the 16MB flash ESP boards like the Wemos D1 mini pro, SDK 2.1.0 add support for it. There is an ongoing PR for the new SDK, and a separate issue for supporting >4MB.
Closing.

closed 2017-10-11 by devyte

mbenitez01, 2019-09-22:
I suffered hesme problems (spiffs seems to geneate infraestructure ok...Arduino shows DATA directory tranered ok..but tfiles aren't found in chip...besides,ftp does not work either.....it happened to me with 2 different modules:  wemos nodemcu and plain 12f without usb.
In the first case, I repeated spiffs generation in Arduino+ DATA transfer several times ...until that at a point everting started to work ok..just MAGICALLYll!
Unfortunatelly, I was not able to discover the cause of this miracle at that time (one month ago).
BUT, tonight, I had been fighting against 12F for a coupe of hours unsuccessfully.. I was not able to load not even my defult html page, when I got iluminated: I generated an  html upload page by code.....stored it in a string varible, and used method from 8266wifiserver 
server.send (200,"text/html",htm_string).
Html upload page generated ok....I was able to upload my index.html page... 
For my total and absolute surprise...when I checked my Filezilla...ALL FILES MAGICALLY APPEARED LOADED IN 8266!  (I'm talking of 40+ files! including images, htmls, js...ad so on.
SO ....LESSON TO SHARE: GENERATE UN UPLOAD WTH THE ANCIENT HTML UPLOAD CONTROL..AND VOILA!! ALL YOUR FILES WILL APPEAR MAGICALLY. (HOPEFULLY YUO ARE AS LUCKYAS I WAS!!!