esp8266/Arduino #2657: ESP8266 crashes during HTTPS request

kpalaniswami, 2016-11-02:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0-rc2

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU 0.9
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		 nodemcu

Decoding 22 results
0x4022b3ff: more_comps at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1072
0x4022b598: alloc at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1106
0x4022b84a: regular_multiply at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 822
0x4020c580: esp_yield at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/core_esp8266_main.cpp line 56
0x4022c3a6: bi_barrett at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1289
0x4022c601: bi_mod_power at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1414
0x4022b464: trim at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1197
0x4022d68d: RSA_public at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 242
:  (inlined by) RSA_encrypt at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 276
0x40229798: send_client_key_xchg at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 352
0x40229b99: do_clnt_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 109
0x4020aafb: ClientContext::_consume(unsigned int) at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
:  (inlined by) ClientContext::read(char*, unsigned int) at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src\include/ClientContext.h line 175
0x40229590: do_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1481
:  (inlined by) basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1357
0x40229938: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 154
0x401004d8: malloc at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x402296e8: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 265
0x4020af6d: SSLContext::connect(ClientContext*, char const*, unsigned int) at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 279
0x4020a332: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x4020b2bd: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 269
0x4020766b: loop at C:\Users\xxxx\Desktop\yyyy/yyyy.ino line 128
0x4020d62c: std::_Function_handler ::_M_invoke(std::_Any_data const&) at C:\Users\xxxx\Desktop\yyyy/yyyy.ino line 128
0x4020c5cc: loop_wrapper at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/cont.S line 109

maurotrochez, 2016-11-10:
Hi, Could you post your code?

kpalaniswami, 2016-11-10:
Fixed the issue myself.

1. Incorrect use of Jsonbuffer – was using it in global scope.
2. readtsringuntil takes up a quite a bit of  time (1sec) to scan HTTP request output.

Thanks for looking at the issue,

Krishnan

From: David Trochez 
Sent: Thursday, November 10, 2016 5:35 AM
To: esp8266/Arduino 
Cc: kpalaniswami ; Author 
Subject: Re: [esp8266/Arduino] ESP8266 crashes during HTTPS request (#2657)

Hi, Could you post your code?

closed 2017-05-08 by igrr