esp8266/Arduino #2658: Unable to update ESP12E trough OTA

Raphyyy, 2016-11-03:
### Basic Infos

#### Hardware
Hardware: ESP12E
Core Version: 2.3.0 (latest)

### Description

Hello
I'm trying to upload the example sketch WebUpdater trough OTA but I'm facing some troubles.
I have no problem to upload the sketch through Serial.
Once it started, I reach my ESP trough my browser (http://172.21.9.138/update)
I choose the .bin just compiled (WebUpdater.ino.bin) and click on upload (like said here http://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html#web-browser)
And then the debug messages on Serial start at "sleep disable" (see above).
So I change ma bauds rate to 115200 and post the Serial Debug above.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		OTA / SERIAL
Reset Method:		ck

### Sketch

```WebUpdater.ino

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>

const char* ssid = "***";
const char* password = "***";

ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;

void setup(void){

  Serial.begin(9600);
  delay(50);
  Serial.println("Booting Sketch...");
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(ssid, password);

  while(WiFi.waitForConnectResult() != WL_CONNECTED){
    WiFi.begin(ssid, password);
    Serial.println("WiFi failed, retrying.");
  }

  httpUpdater.setup(&httpServer);
  httpServer.begin();

  Serial.print("HTTPUpdateServer ready! IP :");
  Serial.println(WiFi.localIP());
}

void loop(void){
  httpServer.handleClient();
}
```

### Debug Messages

```
HTTPUpdateServer ready! IP :172.21.9.138
[... 189 lines cut ...]
3ffffe30:  1a00f8ca a1220040 88a0180c 000fc6c0  
3ffffe40:  10052f70 04080000 001c110c 01010000  
3ffffe50:  05c36730 00000003 f00d2000 16118370  
3ffffe60:  032dfdd8 0000f00d 30e8d761 42673072  
3ffffe70:  e74367d6 30758720 e9167597 fa380064  
3ffffe80:  00000000 00000000 1022a020 a0203360  
3ffffe90:  33271033 1122f004 2230880b 1122f0c0  
3ffffea0:  a0921a0c 11aaf017 1b043237 c02230aa  
3ffffeb0:  0b1122f0 fec95699 427ec882 b847fea0  
3ffffec0:  04323717 1237aa1b 0188901a 7f702a8a  
3ffffed0:  01771005 0d202270 010896f0 2890841b  
3ffffee0:  fffa4601 f041a1a0 f6c611aa 608080ff  
3ffffef0:  f6400800 e4bc6bd5 94821096 143568f9  
3fffff00:  2cbc3d39 dfaa2480 520073bb 209df26a  
3fffff10:  0ca400d0 ebc611aa 052f70ff 0d012210  
3fffff20:  000000f0 67e8a861 47202242 82c44275  
3fffff30:  a620c4e6 72602914 11578020 50401400  
3fffff40:  20509150 b3257060 0000f00d 11427000  
3fffff50:  020c048c 0b114680 b3452054 f00d042d  
3fffff60:  f00d020c 09c0c112 d9e1c9f1 f9c1e9d1  
3fffff70:  0153d6b1 0c607020 70120c06 f0309362  
3fffff80:  c0ff6060 0186fc7c 027d0000 0c0c03fd  
3fffff90:  0c0145d6 d7608e30 23e444b1 8d044217  
3fffffa0:  000201e9 562e72ff 20c02001 55aa55aa  
<<<stack<<<

Raphyyy, 2016-11-04:
EDIT : 
I init ESP8266HTTPUpdateServer with debug = true
Here the log after I choose the example Blink.ino.bin firmware and click on "upload" : 
```
Update: Blink.ino.bin
sleep disable
.............................................................................ERROR[0]: No Error
.ERROR[0]: No Error
.ERROR[0]: No Error
.................................Update Success: 228624
Rebooting...
c_ÇrSöfêÿ‹ƒ††††††
†–jÜ‚‡†’š	Š‹‡X†—ƒ†š
‡†’“‚‚–’š’‹†’‚–ƒ‡‚†‚’†‡‡’’,š‚ƒ‡‚š–‡š‡††‚'“’††–š’‚†ƒ’†‚Š—š––š‡š’–‡‡’††—†š‚Š—’š’‚—‚‚†’žš†Ž‡‹4’’ƒ’‚†–š†ƒˆ‚‚Ž‚‡‚‚Šš–š†›––’,–šŠš–ƒ‚—Ô’†‚š–ŽƒŽš’‚ƒš†‚–†š‚‚ŽX‚*‡‚‚\–ž‡‚‚‚†‚–ž’–š‚‚
Y†‚Žš‚‚‹‚Š–Šš††“†‡‚ƒ’†š’ƒ‚ƒ†Šƒš’ƒ‚š–†š†ƒŠŠÜ–’ƒŠ‚–‚‹’–‚–‚‚‚š‚Šš“–‚†‚Š››Ø‚—Ÿ’š’*ƒŠš‚‡š‚†ƒ‚ƒšš’’ƒŠ††Š‚š’‚†š‚’ƒ›’ƒ‚š’ƒ–ƒ’‚–†‡šÄ‚
’‚š†—š†(‡šš’’‡š‚‡†–š‚–†—�‡†4c_ÇrSøfj�ƒ†††††‚ŠŠ‚Žæ\†‡†’’
Žƒ’Ü†—‹‚
‡šŠ’“†Š’šš’‹†’‚–ƒ‡š†††‚’‚‡‡’—,š‚ƒ‡‚š–‡–ƒ‚‚†/—š‚‚’’–š‚†’’†‚›’–šƒ’–‹‡’††—Šš†‚—’’†—‚‚†’žš††‡‹4’’ƒ’†–š†ƒˆŠ‚†‚ƒ†‚‚š–š†ž–š,–’†’ƒ‚›Ü–†‚Šš’Ž‡Ž’ƒ†…š†–†š††‚X†&‹‚‚T’ž‹†š†‚††’ž’–‚‚Y†‚†‚‡’†‚’‚š‚‚“‚ƒ†ƒ’’†š’†‡‚‚ƒ–šŠ‚’–†š‚‡‚†‚Ô’’ƒ‚†žŠ‹’’‚–’‚‚‚š††[...]
```
Rebooted the ESP, switched to 115200 bauds : 
```
[Troncated chars]
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
```
And it's resetting indefinitely

Anyone ?

Raphyyy, 2016-11-10:
If it can help someone, I found the solution here : http://www.esp8266.com/viewtopic.php?p=57694#p57694

devyte, 2017-10-11:
Per previous comment, closing as resolved.

closed 2017-10-11 by devyte