esp8266/Arduino #2659: EEPROM - does not persist data

sxkod, 2016-11-03:
### Basic Infos

EEPROM write data does not persist over reset. Also the data is not consistent betwee reads.

#### Hardware
Hardware:	ESP-12e on nodemcu 1.0
Core Version: 2.3.0-rc2
Arduino ide 1.6.5-r5

### Description

Hi all

I am struggling to save a tiny amount of data on esp8226 under arduino core! 
I have been banging my head on SPIFFS and this so far declined me. That thread is here

https://github.com/esp8266/Arduino/issues/2655 .

Out of desperation, I have been trying to write to EEPROM instead. So I started with the code that comes with IDE. It works to write and read. However if I try to reset the esp8266 I lose the data. These are the same modules I had used nodemcu-lua and was able to happily create files and store and read back.

I am completely lost now and not sure how to save - may be 64 bytes of data. I even tried different nodemcu 1.0 modules to see if this is just a problem with the particular board. 

Any help or suggestions greatly appreciated.

Thanks

As you can see below, "Will write now" is printed after each reset suggesting the data is not being kept past a reset. 

### Settings in IDE

Module: Generic ESP8266 Module 12e
Flash Size: 4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:	qio?
Flash Frequency:	80Mhz?
Upload Using:	USB
Reset Method:	nodemcu

### Sketch

```
#include <EEPROM.h>

int raddr=0;
int cnt=0;
int waddr=0;

void setup() {
Serial.begin(115200);
EEPROM.begin(512);
delay(100);
if ((byte)EEPROM.read(0)!=8){
  Serial.println("Empty. Will write now");
  delay(10);
  for (int n=0;n<10;n++){
    EEPROM.write(waddr,(byte)8);
    waddr++;
    delay(100);
    }
  EEPROM.commit();
}
EEPROM.end();
waddr=0;
}

void loop() {
  EEPROM.begin(512);
  delay(100);
  Serial.println("");
  cnt++;
  raddr=0;
  if (cnt>10) {ESP.restart();}
  
  for (int n=0;n<10;n++){
    Serial.print(EEPROM.read(raddr));
    Serial.print('\t');
    delay(100);
    raddr++;
  }
  EEPROM.end();
  delay(200);
}
```

### Debug Messages
As you can see below not all the reads are the same, so it's not consistent. And the values don't seem to keep over reset.
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
�Empty. Will write now

119	1	0	0	8	8	8	8	8	8	
21	0	0	0	244	9	255	63	42	0	
21	0	0	0	244	9	255	63	42	0	
14	2	21	0	244	9	255	63	42	0	
21	0	0	0	244	9	255	63	42	0	
21	0	0	0	244	9	255	63	42	0	
14	2	21	0	244	9	255	63	42	0	
21	0	0	0	244	9	255	63	42	0	
21	0	0	0	244	9	255	63	42	0	
14	2	21	0	244	9	255	63	42	0	

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
�Empty. Will write now

119	1	0	0	8	8	8	8	8	8	
21	0	0	0	244	9	255	63	42	0	
21	0	0	0	244	9	255	63	42	0	

```

WereCatf, 2016-11-04:
Works fine for me, no issue at all. Try setting flash-mode to DIO and frequency to 40MHz instead and see if that fixes it.

sxkod, 2016-11-07:
Hi WereCatf

Thanks for your reply. I tried changing it to other, bare boards - I mean esp12e directly, and changed all the settings in the IDE. Unfortunately no luck. I solved my problem by adding an external at24c32 I have lying around. Still would have been nicer if I was able to use the internal EEPROM rather than having to add components. I am not sure where things went wrong!

Thanks

samakar, 2016-12-18:
Hi sxkod,
I tested your code on WROOM-2  (flash at 40MHz and CPU at 80Mhz). It works without issue. WROOM-2 is powered through usb. Then I changed `for (int n=0;n<10;n++)`  to `for (int n=0;n<12;n++)`
Then the last two digits became random. So I confirm the behavior you saw though it may happen at different values of n.
I revised your code by primarily removing all the delays. It works very well now with 'Serial.flush();' commented out or not.  Seems extra delays hurt. Maybe some internal routines find time to kick in!

```
#include <EEPROM.h>

int raddr=0;
int cnt=0;
int waddr=0;
const int MAX_N = 50;
const byte NUMBER = 32;
const int SIZE = 512;

void setup() {
Serial.begin(115200);
delay(1000);
EEPROM.begin(SIZE);
if ((byte)EEPROM.read(0)!=NUMBER){
  Serial.println("Empty. Will write now the number " + NUMBER);
  //Serial.flush();
  for (int n=0;n<MAX_N;n++){
    EEPROM.write(waddr,NUMBER);
    waddr++;
  }
}
EEPROM.end();
waddr=0;
}

void loop() {
  EEPROM.begin(SIZE);
  Serial.println("");
  //Serial.flush();
  cnt++;
  raddr=0;
  if (cnt>10) {ESP.restart();}
  
  for (int n=0;n<MAX_N;n++){
    Serial.print(EEPROM.read(raddr));
    Serial.print(' ');
    raddr++;
  }
  //Serial.flush();
  EEPROM.end();
  delay(200);
}
```

sxkod, 2016-12-19:
Thanks Samakar. I will try the way you did and see if that helps.

sxkod, 2016-12-19:
I tested it with a few different variations, such as writing with incrementing NUMBER, printing as a char to confirm the incrementing number, writing the same number over and over etc. 

It works. Thanks a lot for posting the solution. Not sure what the problem is though :-(

closed 2016-12-19 by sxkod