esp8266/Arduino #2661: DeepSleep not working

DoinStuff35, 2016-11-04:
### Basic Infos

#### Hardware
Hardware:			ESP-12 E, DHT-22, LD1117V33, 3,7V Li-Po 3000mAh
Core Version:      	current

### Description

Despite DeepSleep constantly draws 80mA, DeepSleep seems not to work.

### Sketch

```cpp

#include <DHT.h>
#include <ESP8266WiFi.h>

// replace with your channel’s thingspeak API key,
String apiKey = "******************************";
const char* ssid = "**********";
const char* password = "*********************";

const char* server = "api.thingspeak.com";
#define DHTPIN 12 // what pin we’re connected to

DHT dht(DHTPIN, DHT22, 15);
WiFiClient client;

void setup() {

  Serial.begin(115200);
  delay(10);
  dht.begin();

  WiFi.mode(WIFI_STA); // added 300716
  WiFi.begin(ssid, password);

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");

  float h = dht.readHumidity();
  float t = dht.readTemperature();

  if (isnan(h) || isnan(t)) {
    Serial.println("Failed to read from DHT sensor!");
    return;
  }

  if (client.connect(server, 80)) { // "184.106.153.149" or api.thingspeak.com
    String postStr = apiKey;
    postStr += "&field3=";
    postStr += String(t);
    postStr += "&field4=";
    postStr += String(h);
    postStr += "\r\n\r\n";

    client.print("POST /update HTTP/1.1\n");
    client.print("Host: api.thingspeak.com\n");
    client.print("Connection: close\n");
    client.print("X-THINGSPEAKAPIKEY: " + apiKey + "\n");
    client.print("Content-Type: application/x-www-form-urlencoded\n");
    client.print("Content-Length: ");
    client.print(postStr.length());
    client.print("\n\n");
    client.print(postStr);

    Serial.print("Temperature: ");
    Serial.print(t);
    Serial.print(" degrees Celcius Humidity: ");
    Serial.print(h);
    Serial.println("% send to Thingspeak");
  }
  client.stop();

  Serial.println("Waiting…");

  delay(500);

  WiFi.disconnect();

  // thingspeak needs minimum 15 sec delay between updates

  //wifi_set_sleep_type(LIGHT_SLEEP_T);

  //delay(900000);

  Serial.println("ESP8266 in sleep mode");

  delay(500);
  
  ESP.deepSleep(60 * 15 * 1000000); // 15 Minuten

  delay(1000);

}

void loop() {

}
```

dreamfalcon, 2016-11-04:
Are you using one of those white boards for ESP-12?
If yes check this video at 10:12.

DoinStuff35, 2016-11-04:
Nope, i don´t use any Resistor. And...he´s talking about **micro** Ampere..i´m talking about **milli** Ampere. Anyway, thanks for trying to help!

Picture of my "Project" added.
[image]

supersjimmie, 2016-11-05:
Interesting, I have such a setup with an esp01 and simmilar code (same idea).
My battery drains much faster than it should. I only send data every hour and yet it only lasts one week.  I don't have a working multimeter (broken, no new one yet) so I am currrently unable to measure. Seeing your post makes me think that I too have that problem.

One thing is that you might better use L after each number in your sleeptime calculation, like 60L * 15L * 1000000 so it does not try to use 60 and 15 as an int, because that would cause a very short deepsleep.
Have you tried a Serial.print("test"); after the deepsleep to be sure it never reaches there?
And perhaps you can see on your wifi accesspoint/router if it really disappears?

DoinStuff35, 2016-11-05:
Hey Jimmie,

thanks for your input! I feel a little stupid right now:

I´ve inserted your "test" and DeepSleep really seems to work (no "test" on console). Even added "L" in sleeptime. Still 80mA! At last i´ve remembered Mr. Ohm and his law: I´ve used a 3,7 LiPo and a Multimeter to measure the current...doesn´t have a Multimeter a resistor too? Switched to more powerful energy source and, you won´t believe it, **80mA for a few seconds (connecting wifi), drops to 5,3 mA (sleep, maybe LD1117 quiescent current).**

So on a 3.000 mAh LiPo i´ll reach out foralmost a month of uptime. Still not reaching my goal (half a/year), but far better.

So, i think this issue can be marked "closed", although i really like to discuss more options to lower the current consumption...

supersjimmie, 2016-11-06:
I would think 5.3mA is still not normal with the esp in deepsleep, an lm1117 and a dht22. Afaik the dht22 has a very low standby current and the lm1117 is also very efficient, so together they shouldn't draw a constant 5.3mA. Perhaps you can test the current through an lm1117 without any load and then with only the dht22, but still without the esp?  If only the lm1117 with the dht22 then still draw nearly 5.3mA, the esp in deepsleep is not the cause and works as expected. If they draw much less, the esp draws the difference.

DoinStuff35, 2016-11-06:
Interesting:

5.24 - 5.27 mA | LD1117 only
5,26 - 5.28 mA | LD + DHT
5.31 - 5.35 mA | LD + DHT +ESP (in DeepSleep)

The LD seems to be the culprit...maybe i´m giving the XC6203E332PR a shot.

supersjimmie, 2016-11-07:
Thanks, I think that explains it for me too.
I wonder how other people (also using a battery) are doing this?

devyte, 2017-10-11:
Closing as resolved.

closed 2017-10-11 by devyte