esp8266/Arduino #2663: Using GDB for online debugging ESP8266

W2K, 2016-11-05:
### Basic Infos

#### Hardware
Hardware:			ESP8266 (NodeMCU)

### Description

Problem description

For couple of days I've been struggling to configure debugging for ESP8266 running on Arduino and using PlatformIO.
I managed to build a proper binary file with http://blog.squix.org/2016/04/esp8266-offline-debugging-with.html:
1. Added build_flags = -Og -ggdb -DDEBUG_ESP_PORT=Serial to platformio.ini
2.Downloaded full xtensa SDK as this provided for PIO does not include xtensa GDB
Thanks to these steps I am able to offline debug - see lines numbers etc,but I cannot make it work with online debugging. Here is what i ve done:
4. Added GDBStub.h include to my mian class
5. Started GDB with firmware that has debugging symbols included(form step 1) to debug via COM - set baud rate to 115200 (the same as I use for normal state check via serial moniotr - wifi messages etc.) .
6. I get no response - only timouts etc
7. In the result I would like to use CLion to insert breakpoints watches etc.

Can someone pint me what I'm mising ? I have 2 suspicions : 
* Is inluding only GDBAtub.h enough or shoul I call gdb_init somewhere (its's not recognized as a function) or do anything more.
* Baud rate - should I use the same as the one set for serial communication messages (115200) ?

I am using Windows.

After some reading I  changed  build flags to : 

build_flags = -Og -ggdb -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_TLS_MEM

Module:  			NODEMCUv2 / ESP12E
### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ArduinoOTA.h>
#include <pb_encode.h>
#include <pb_decode.h>
#include <ESP8266mDNS.h>
#include <GDBStub.h>

void setup() {
 Serial.begin(115200);
}

void loop() {

}
```

### Debug Messages

```
messages here
```

igrr, 2016-11-07:
GDBStub will be invoked when exception happens, as in its current configuration it supports only post-mortem debugging. GDBStub has a feature to break into debugger on Ctrl-C (in uart console, not in GDB), but it has to be activated in GDBStub, and it may conflict with HardwareSerial which also installs its own UART interrupt handler. I once added a "break into debugger on button" hack to GDBStub, it requires replacing UART interrupt with a GPIO interrupt. You can try that if you like.

tomaszchacuk, 2016-11-07:
Thanks for reply.  I was thinking of kind of debugging that does not happen on some kind of exception but rather possibility investigate live-state  of variables. Preferably I would like to achieve an effect like described in this tutorial for Sming( Eclipse/CLion integration etc):  https://blog.attachix.com/live-debugging-with-open-source-tools-programming-for-esp8266-part-4/

freeck, 2016-12-12:
I cannot follow all the details and implication of the discussion above, but I observe that from version 1.6 on, debugging isnt functioning anymore. 
May we expect a solution to this issue soon? Or more detailed documentation to solve this on application level?

freeck, 2017-06-05:
I solved my issue see: https://github.com/Sloeber/arduino-eclipse-plugin/issues/593
However os_install_putc1() does not seem to work in release 2.3.

devyte, 2017-10-11:
@W2K is this issue still valid? Have you tried latest git?

freeck, 2017-10-11:
Same issue using 2.4.0rc1, (is that the latest git?) but as reported earlier in this thread my issue can be closed.

closed 2017-10-11 by devyte