esp8266/Arduino #2664: Can not download esp8266 board manager url (again)

davolesh, 2016-11-06:
Similar to Issue [https://github.com/esp8266/Arduino/issues/1689](url)

[image]

When manually downloading both from domain and Ip:

[image]

I tried from multiple networks, including cellular and same issue reoccurs. Any help would be highly appreciated, thanks!

hashin, 2016-11-14:
I am getting this issue again. Any possible reason? @igrr 

Please update @davolesh

davolesh, 2016-11-14:
I figured out the issue for me. I have OpenDNS on my network. So I got the free version of TunnelBear and once i got that running the setup went through flawlessly! It must have been a DNS issue. Give it a shot. Either disable any DNS or try a VPN.

davolesh, 2016-11-14:
Let me know if this works for you and if I can close the issue.

hashin, 2016-11-15:
Working for me now. You can close the issue. :+1:

closed 2016-11-15 by davolesh