esp8266/Arduino #2666: Error compiling Blink example

zfred68, 2016-11-06:
This is probably a setup problem. 
On Fedora 24, built arduino and esp8266 from git clone
ant run
or exec'ing arduino from installing the resultant arduino-1.6.13-linux64.tar.xz

===========
Arduino: 1.6.13 (Linux), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

/home/fred1/bin/arduino/arduino-1.6.13/arduino-builder -dump-prefs -logger=machine -hardware /home/fred1/bin/arduino/arduino-1.6.13/hardware -tools /home/fred1/bin/arduino/arduino-1.6.13/tools-builder -tools /home/fred1/bin/arduino/arduino-1.6.13/hardware/tools/avr -built-in-libraries /home/fred1/bin/arduino/arduino-1.6.13/libraries -libraries /home/fred1/Arduino/libraries -fqbn=esp8266com:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10613 -build-path /tmp/arduino_build_994617 -warnings=more -prefs=build.warn_data_percentage=75 -verbose /home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/libraries/esp8266/examples/Blink/Blink.ino
/home/fred1/bin/arduino/arduino-1.6.13/arduino-builder -compile -logger=machine -hardware /home/fred1/bin/arduino/arduino-1.6.13/hardware -tools /home/fred1/bin/arduino/arduino-1.6.13/tools-builder -tools /home/fred1/bin/arduino/arduino-1.6.13/hardware/tools/avr -built-in-libraries /home/fred1/bin/arduino/arduino-1.6.13/libraries -libraries /home/fred1/Arduino/libraries -fqbn=esp8266com:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10613 -build-path /tmp/arduino_build_994617 -warnings=more -prefs=build.warn_data_percentage=75 -verbose /home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/libraries/esp8266/examples/Blink/Blink.ino
Using board 'generic' from platform in folder: /home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266
Using core 'esp8266' from platform in folder: /home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266
Detecting libraries used...
"/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/sdk/lwip/include" "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_994617/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10613 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/variants/generic" "/tmp/arduino_build_994617/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/sdk/lwip/include" "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_994617/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10613 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/variants/generic" "/tmp/arduino_build_994617/sketch/Blink.ino.cpp" -o "/tmp/arduino_build_994617/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/fred1/bin/arduino/arduino-1.6.13/tools-builder/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_994617/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/sdk/lwip/include" "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_994617/core" -c -Wall -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10613 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/fred1/bin/arduino/arduino-1.6.13/hardware/esp8266com/esp8266/variants/generic" "/tmp/arduino_build_994617/sketch/Blink.ino.cpp" -o "/tmp/arduino_build_994617/sketch/Blink.ino.cpp.o"
as: unrecognized option '--text-section-literals'
exit status 1
Error compiling for board Generic ESP8266 Module.
===========

I am suspicious of " -tools /home/fred1/bin/arduino/arduino-1.6.13/hardware/tools/avr "
that can't be right.
I'm really just started with arduino for the ESP8266, so not very familiar with it.
can't work out how the ide itself is directed to either avr or esp8266, where does it
get that information from?

looked at preferences.txt

maybe relevant entries...

board=generic
boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json
....
target_package=esp8266com
target_platform=esp8266
....

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			generic ESP8266
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

is there a means of printing out the IDE settings? - can't find any menu

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

standard Blink example
### Debug Messages

```
messages here
```

> 

[Uploading preferences.txt…]()

gorghino, 2016-11-08:
Hi @zfred68. Which project homepage's procedure are you using? Building with make?
Did you try to use Arduino IDE instead?

zfred68, 2016-11-08:
Hi @gorghino . The above is the result of building with git clone the repos, and make procedure. so i would need to know how the ide works a bit to solve that. 
I know the java source code is there, but I'm not sure I want to invest in that time, as my fav build environment is eclipse and i have found the arduino eclipse plugin and have that working now with esp8266. With avr,I have always just programmed in C and used stk500 for programming the chips,
so have never got into this arduino method before.
I did as well,install the arduino ide from the fedora rpm repository, but it is without the esp8266.

with the arduino git repo, it is also inconvenient that the build.xml is insufficient for eclipse to import 
as an ant java project, which means i have to hand write a project xml file, thats just unbearable for me.

gorghino, 2016-11-14:
Hi @zfred68 . I've always used Arduino IDE with esp8266 Arduino library so I can't help you if you're trying to compile it for Eclipse :( 
Maybe some devs could help you.

zfred68, 2016-11-17:
sorry, I must correct your misunderstanding. The above problem is about arduino, not eclipse.
ie, installing arduino from source with 
git clone https://github.com/arduino/Arduino.git

zfred68, 2016-11-25:
looks like arduino ide is not invoking the xtensa-lx106-elf-as, for .s but rather the system default as
i re-did the command in a cmd prompt with added option -save-temps -v
so in the relevant /tmp/../sketch dir
if I do invoke the 
arduino-1.6.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-as Blink.ino.s -o Blink.ino.cpp.o

the object file is produced with no complaint,

I checked the platform.txt file in the  hardware/esp8266com/esp8266/platform.txt

...
compiler.as.cmd=xtensa-lx106-elf-as
...
compiler.as.cmd=xtensa-lx106-elf-as
..

so why?

elijah-taylor, 2017-05-13:
I ran into this same issue I think.  AFAICT the xtensa toolchain that is distributed for platformio is broken/inconsistent between archs and releases.

In my particular case https://dl.bintray.com/platformio/dl-packages/toolchain-xtensa-linux_armv6l-1.40802.0.tar.gz is broken.  xtensa-lx106-elf-c++ was directly invoking system |as|, and the tarball also had broken symlinks (the symlinks were relative to the tarball root and not relative).  This leads me to believe this package was packed (maybe repacked(?) because of different user/group than the other bundles) incorrectly, maybe in a chroot with it as the default toolchain or something.

My workaround was to unpack the contents of https://dl.bintray.com/platformio/dl-packages/toolchain-xtensa-linux_armv6l-2.tar.gz on the toolchain directory instead.

elijah-taylor, 2017-05-13:
opened an issue specifically for armv6 on the platformio tracker, whose issue I think this is: https://github.com/platformio/platform-espressif8266/issues/33

devyte, 2017-10-11:
Assumin the issue is still valid, I don't see an issue with the code in this repo in the descriptions posted here. Arduino IDE 1.8.4 works fine under Linux, and building the core from git also works fine.
Closing as can't reproduce. If the problem is still valid, please open a new issue and add all the relevant info.

closed 2017-10-11 by devyte