esp8266/Arduino #2669: Exception: InstructionFetchErrorCause

alkopop79, 2016-11-07:
### Basic Infos
Mac OS X 10.11.6
Arduino IDE 1.6.9

#### Hardware
Hardware:		ESP8266
Core Version:      	-

### Description

pic of the module: https://cdn.sparkfun.com//assets/parts/1/1/1/2/9/13678-01.jpg

whichever sketch I try, I get the same error message on serial (see: stack trace). The module is flashed using a USB-to-serial (3V) adaptor with the following wiring:

TX -> RX
RX -> TX
HIGH -> CH_PD
LOW -> GPIO0
LOW -> HIGH -> LOW - RESET (the pin is pulled low then high to put module in flash mode)

The module is powered from the USB adaptor.  The IDE shows me clearly that the flash is written. I have already tried Exception Decoder and got the following results:

```
Decoding 85 results
0x40101b01: RC_GetBlockAckTime at ?? line ?
0x40101afe: RC_GetBlockAckTime at ?? line ?
0x402067ec: esf_buf_alloc at ?? line ?
0x40101b07: RC_GetBlockAckTime at ?? line ?
0x40101ae8: RC_GetBlockAckTime at ?? line ?
0x402016fc: __wrap_system_restart_local at /Users/alkopop79/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_postmortem.c line 190
0x40206538: wDevEnableRx at ?? line ?
0x4022ebf0: pm_wakeup_init at ?? line ?
0x40206538: wDevEnableRx at ?? line ?
0x4020661b: wdev_go_sniffer at ?? line ?
0x40206538: wDevEnableRx at ?? line ?
0x40101b07: RC_GetBlockAckTime at ?? line ?
0x40101ae8: RC_GetBlockAckTime at ?? line ?
0x40101b01: RC_GetBlockAckTime at ?? line ?
0x40101afe: RC_GetBlockAckTime at ?? line ?
0x402067ec: esf_buf_alloc at ?? line ?
0x40101b07: RC_GetBlockAckTime at ?? line ?
0x40101ae8: RC_GetBlockAckTime at ?? line ?
0x40101b01: RC_GetBlockAckTime at ?? line ?
0x40101afe: RC_GetBlockAckTime at ?? line ?
0x402067ec: esf_buf_alloc at ?? line ?
0x40101b07: RC_GetBlockAckTime at ?? line ?
0x40101ae8: RC_GetBlockAckTime at ?? line ?
0x40101b01: RC_GetBlockAckTime at ?? line ?
0x40101afe: RC_GetBlockAckTime at ?? line ?
0x402067ec: esf_buf_alloc at ?? line ?
0x40101b07: RC_GetBlockAckTime at ?? line ?
0x40101ae8: RC_GetBlockAckTime at ?? line ?
0x40101b01: RC_GetBlockAckTime at ?? line ?
0x40101afe: RC_GetBlockAckTime at ?? line ?
0x402067ec: esf_buf_alloc at ?? line ?
0x40101b07: RC_GetBlockAckTime at ?? line ?
0x40101ae8: RC_GetBlockAckTime at ?? line ?
0x40101b01: RC_GetBlockAckTime at ?? line ?
0x40101afe: RC_GetBlockAckTime at ?? line ?
0x402067ec: esf_buf_alloc at ?? line ?
0x40101b07: RC_GetBlockAckTime at ?? line ?
0x40101ae8: RC_GetBlockAckTime at ?? line ?
0x40101b01: RC_GetBlockAckTime at ?? line ?
0x40101afe: RC_GetBlockAckTime at ?? line ?
0x402067ec: esf_buf_alloc at ?? line ?
0x40101b07: RC_GetBlockAckTime at ?? line ?
0x40101ae8: RC_GetBlockAckTime at ?? line ?
0x40101b01: RC_GetBlockAckTime at ?? line ?
0x40101afe: RC_GetBlockAckTime at ?? line ?
0x402067ec: esf_buf_alloc at ?? line ?
[... 300 lines cut ...]
3ffffe40:  ffffffff ffffffff ffffffff ffffffff  
3ffffe50:  ffffffff ffffffff ffffffff ffffffff  
3ffffe60:  ffffffff ffffffff ffffffff ffffffff  
3ffffe70:  ffffffff ffffffff ffffffff ffffffff  
3ffffe80:  ffffffff ffffffff ffffffff ffffffff  
3ffffe90:  ffffffff ffffffff ffffffff ffffffff  
3ffffea0:  ffffffff ffffffff ffffffff ffffffff  
3ffffeb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffec0:  ffffffff ffffffff ffffffff ffffffff  
3ffffed0:  ffffffff ffffffff ffffffff ffffffff  
3ffffee0:  ffffffff ffffffff ffffffff ffffffff  
3ffffef0:  ffffffff ffffffff ffffffff ffffffff  
3fffff00:  ffffffff ffffffff ffffffff ffffffff  
3fffff10:  ffffffff ffffffff ffffffff ffffffff  
3fffff20:  ffffffff ffffffff ffffffff ffffffff  
3fffff30:  ffffffff ffffffff ffffffff ffffffff  
3fffff40:  ffffffff ffffffff ffffffff ffffffff  
3fffff50:  ffffffff ffffffff ffffffff ffffffff  
3fffff60:  ffffffff ffffffff ffffffff ffffffff  
3fffff70:  ffffffff ffffffff ffffffff ffffffff  
3fffff80:  ffffffff ffffffff ffffffff ffffffff  
3fffff90:  ffffffff 450eca59 a39f7846 20fc6218  
3fffffa0:  000201e9 1a839a3f ffffff00 55aa55aa  
<<<stack<<<
```

pipi61, 2016-11-07:
use:   while(true) yield();

alkopop79, 2016-11-07:
Thanks pipi61, will give it a try!

alkopop79, 2016-11-07:
Didn't help. It must be h/w problem. Any ideas?

pipi61, 2016-11-07:
power stable? use elko on power pin?
[https://github.com/esp8266/Arduino/blob/master/doc/boards.md](url)
Improved Stability

alkopop79, 2016-11-07:
Sorry pipi61, 404 :(

lrmoreno007, 2016-11-07:
You have Reset cause 2 ---> rst pin

You must pull up RST pin.

alkopop79, 2016-11-09:
Reset pin is pulled up. I have tried another board (looks like a different make) and the same thing happened.

lrmoreno007, 2016-11-09:
On the other hand, you want "provoke the s/w wdt firing..." and it is what you have ...

Where is the problem?

devyte, 2017-10-11:
@alkopop79 assuming the issue is still valid, either you have a flash layout issue, or your power is not enough, or your upload speed is too high for your flash which corrupts the writing.
In any case, the current core doesn't have any issues with flashing the ESP01 et al. If you still need assistance, please refer to a community forum to discuss what is wrong with your setup.
Closing, see #3655 .

closed 2017-10-11 by devyte