esp8266/Arduino #2671: Basic example for modified ethernet.h

FernandoGarcia, 2016-11-07:
Hi !

Is possible give us a working basic example for this modified ethernet.h library?
The default examples available here doesn't work without changes.
https://github.com/esp8266/Arduino/tree/master/libraries/Ethernet/examples
Best regards.

Humancell, 2016-11-08:
Also ... is there a clear document or diff that can show exactly what was changed in this version of the lib to work with the ESP?

I'm wanting to use both the W5100, and the better/faster W5500 for Ethernet, and so I want to update this support to support both ... but I'm hoping I don't have to start from scratch and guess?

neu-rah, 2016-11-08:
no docs yet, v3 (dev branch ) was a rewrite, and not just because of esp8266, so a lot has changed from 2.x.. the good news it's that i'm already started from scratch :D

Humancell, 2017-05-22:
Curious if anyone has any experience using this with the W5500?  I'm digging in to see if I can get it working, but was curious if anyone knows the pitfalls I might run into?

devyte, 2017-10-11:
@Humancell did you make any progress? if so, you may want to open a new issue as a feature request, and make a PR.

devyte, 2017-10-11:
@FernandoGarcia There are examples provided together with the lib.
Closing.

FernandoGarcia, 2017-10-11:
You should check my issue firts!
As I told the examples doesn't work.

https://github.com/esp8266/Arduino/tree/master/libraries/Ethernet/examples

devyte, 2017-10-11:
@FernandoGarcia sorry, I took your post to mean you were requesting examples.
You're sayying the current examples don't work with the 510?

FernandoGarcia, 2017-10-11:
Look for examples and tell me where to put SSID and password at least. It was just a copy and paste from Arduino IDE

d-a-v, 2018-05-07:
@Humancell since you have W5100 (which I don't have),
can you try https://github.com/d-a-v/W5500lwIP with both W5100 and W5500 ?

PR #6812 (2020-07-16): replace old Ethernet Arduino-library with v2, submoduled
supports W5500, W5200 and W5100

*edit*:
closes #2671 
fixes #3943
closed 2020-07-16 by d-a-v