esp8266/Arduino #2672: ADC Values Inaccurate and different from NodeMCU

lawrence-jeff, 2016-11-08:
Hardware:	Multiple NodeMCU boards (Amico and Lolin) (with ESP-12 AIThinker module)
Core Version:      	2.3.0
### Description
The ADC Results via analogread (TOUT) don't seem to be correct, not wildly inaccurate but enough to throw some sensors.
Without changes to the setup they are also different than what is reported by the nodemcu lua firmware (which seems more accurate)
Even simple tests show it
Compare
Measured/Calculated values on Tout pin (factoring in voltage divider in nodemcu boards) and measured by a decent meter

1.02v (Pin fed via 3.32 source) - Nodemcu reads 1024  Arduino 1002
.502v - Nodemcu reads 530, Arduino 485
0 v  - Both read 1

I can replace devices and reflash and reboot and the measurements are consistent. I never see the Arduino go to 1024 even when it has over 1 volt on the pin which seems very odd. Nodemcu firmware reports the correct value of 1024 consistently

Any ideas?

igrr, 2016-11-08:
Can you try adding the following line somewhere before your `setup` function?

```
ADC_MODE(0);
```

lawrence-jeff, 2016-11-08:
I added it and it didn't result in any noticeable change. Measuring the pin at 1.03 still results in around 1002 in Arduino and 1024 in nodemcu (what I would expect if the range is 0-1v)

Let me know if there is anything else I should look at.
Jeff

lawrence-jeff, 2016-11-09:
lrmoreno007,
Not sure exactly what you are asking - I'm using the arduino software not a physical board and the Arduino ESP port seems adjusted to measure 0-1v which is correct for an ESP ADC.

The repro process is pretty basic
Setup an ESP with 1 volt to the ADC pin (on a nodemcu board this means connect the A0 pin to the 3.3v regulated output which results in 1.03 volts at the module pin (measured)
This should result in a 1024 value in all software since it is at the top of the 1v range

Now flash with Nodemcu Lua and read the ADC = you get 1024 as expected
Don't touch any hardware but flash with arduino port 
Do an Analogread and you get ~1000

Not a huge difference in this example but doing the same test at other levels shows larger discrepancies between the analog reads. Whatever Nodemcu Lua does seems to provide more accurate numbers

Not exactly relevant but if you use a real arduino board with the same code (and use the Aref pin and connect to the 3.3v rail) and run the same code you also get 1024 as expected.

So it doesn't seem to be the hardware or a problem inherent in the Arduino core itself - but something in the ESP Arduino port.

lawrence-jeff, 2016-11-09:
Also interesting that Issue #338 while not focused on this component noted the same thing in the initial report - that person was getting 909 in the arduino port and 1024 in nodemcu. The issue focused more on why the grounded value was wrong but it seems they were also seeing the general discrepency.

lrmoreno007, 2016-11-09:
Ok, i think you was measuring with a Arduino Hardware. I doesn't read you was using NodeMcu Lua firmware.

lrmoreno007, 2016-11-09:
Can you do the following? And tell us the result.

`ADC_MODE(ADC_VCC);

void setup() {
  float voltaje = ESP.getVcc();
  Serial.print(voltaje/102.40f);
  Serial.println(" <--- Must be 33");
}

void loop() {

}`

lawrence-jeff, 2016-11-09:
32.71 <--- Must be 33

lrmoreno007, 2016-11-09:
In Vcc reading mode everything seems fine and the applied voltage is adequate.

Sorry to disagree with Igrr but the line before setup should not be? ... 
ADC_MODE(ADC_TOUT);

Or

ADC_MODE(ADC_TOUT_3V3);

Or

ADC_MODE(33);

Esp.h
' enum ADCMode {
    ADC_TOUT = 33,
    ADC_TOUT_3V3 = 33,
    ADC_VCC = 255,
    ADC_VDD = 255
};

#define ADC_MODE(mode) int __get_adc_mode(void) { return (int) (mode); }'

lawrence-jeff, 2016-11-09:
So I tried both ADC_MODE(33) and ADC_MODE(255)
33 still results in 1002 instead of 1024 and 255 results in 65535

So is this something with only my boards - since it such an easy test does the max value on the pin result in 1024 as expected for everyone else?

lrmoreno007, 2016-11-16:
Sorry but so far I have not had any NodeMCU free.

I have tested with NodeMCU and Arduino and both give me the same result.

lawrence-jeff, 2016-11-19:
Well your Lua code agrees with mine but when I run that same arduino code (with a direct jumper between A0 and 3.3v) and I get 
1001.00
1001.00
1001.00
1001.00

I have tried 4 different nodemcu boards so far, all similar.
Since you declare it as a float why does your output not have a decimal component? What version of the arduino esp8266 are you using. I am using 2.3.0

lawrence-jeff, 2016-11-19:
Followed the GIT instructions to load the latest and still get 1001.00.

lawrence-jeff, 2016-11-19:
Reproduced on another system - using ESParduino 2.3.0 the top value is 1002 or so, replaced the contents of the 2.3.0 folder with the latest from this repo and recompiled and the top is now 1024 so one problem is fixed. The midpoint value is still off 

Arduino and LUA both read the exact midpoint (validated with meter on the pin as .501v) as 530 instead of 512. If I take a real arduino  add to the same circuit (using nodemcu board power rails) and use the 3.3v off nodemuc as external VREF on the real arduino it measures the same point as 510 (which would be basically correct as 1024/2 = 512)

So better but still oddly off ...since its present in both LUA and ESPArduino probably something in the underyling SDK, so I guess this can be closed

jourjine, 2016-11-19:
maximum voltage is 1.06 not 1.01
rawADC 978 = 1v
etc

about [this](http://www.esp8266-projects.com/2015/03/internal-adc-esp8266.html)

lawrence-jeff, 2016-11-20:
jourjine - I read through that article but I don't think it explains what I am seeing on my modules. The original problem was that the ESPArduino Analog read seemed to top out at around 1000 even when the same circuit in LUA measured 1024. According to the datasheet the range is 0-1v so it should have been 1023/1024 like LUA. This issues is resolved by using the GIT version of the ESP arduino code and now I have a top of 1024. From measurement it seems though that this top is reached now at around .97 volts  - so it has gone too far the other way.  What I did was use a pot to adjust voltage on the pin for every .1 v from 0-1v. After a bit of math dividing the ADC value by 1055 seems to correct it to reflect reality in volts. So if my ADC reads 800 that translates to .76 volts. If it reads 1023 that translates to .97 volts. This seems to give me nice results that work across the 3 boards i have.

If yours is working - be careful about compiling or changing your code - you could find the behavior changes more than you would expect.

closed 2016-12-03 by lawrence-jeff

universam1, 2016-12-29:
Hi I can confirm that this problem exists with the 1.5.3 SDK and is gone with the GIT version of Arduino with SDK 1.54!
Any explanation for this?

davidegironi, 2017-04-24:
Hello, I've the same issue here.
NodeMCU v3 board. It has a voltage divider formed by 220k and 100k resistor attached to A0.
A0 is connected to 3.272 board output.
analogRead(A0) -> 978raw value, 1.02V read with a multimeter on A0 (which is a right voltage value according to the voltage divider), what's wrong is the ADC raw value.
Two different NodeMCU v3 board, and one NodeMCU Amica tested, same results.
ESP.getVcc() return 2.7V almost, on each of the 3 boards tested, it sounds strange.
Compiled and uploaded using Windows or Linux, doesn't make difference.
WiFi on or off doesn't make difference.
Thanks for help.

brianjmurrell, 2017-12-21:
I too am having issues with inaccurate ADC values on my ESP board and Arduino.  Looking in the board manager in the Arduino IDE it seems I also have version 2.3.0.

Given that this issue is over a year old, why has a new version of the software not been released to fix this problem (probably a rhetorical question in this issue).

lawrence-jeff, 2017-12-21:
I would open a new issue (this one is closed), if you can get better results using NodeMCU Lua firmware that might help get attention

brianjmurrell, 2017-12-21:
@lawrence-jeff Much thanks.  I already did.  #4009.  :-)

> if you can get better results using NodeMCU Lua firmware 

That would be interesting for sure.  But this is all pretty new to me.  I jumped immediately to using Arduino with it.  Any docs/guides/advise you can give about how I would test this with NodeMDU Lua firmware now?