esp8266/Arduino #2673: SimpleAuthentification is completely insecure

aykevl, 2016-11-08:
The [SimpleAuthentication example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/SimpleAuthentification/SimpleAuthentification.ino) is completely insecure. The only thing needed to 'authenticate' is setting the cookie to a known and public value, which is like buying a safe and putting the code in the manual lying besides it. Not secure at all.

Well, there are actually a few issues:

1. As noted above.
2. The password is a variable buried in the source code - it should ideally be at the top with a note the password should be changed.
3. The password is not checked using a constant-time string comparison, making it possible to recover the password (see https://codahale.com/a-lesson-in-timing-attacks/).

I would suggest either removing it (so people at least don't get a false sense of security) or implementing a proper form of authentication. At the very least, I would expect a comment describing these issues so people don't get a wrong sense of security or learn wrong approaches to security.

me-no-dev, 2016-11-08:
it's called SimpleAuthentication and not SecureAuthentication.
To have reliable authentication over unsecured channel would require resources way above what the ESP offers. If you have an idea, feel free to submit a pull request.

roboticboyer, 2016-11-18:
You can also take a look to [https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HttpBasicAuth/HttpBasicAuth.ino](url)

devyte, 2017-10-10:
@aykevl you are correct, it is not secure. However, it is part of HTTP, and known to be not secure. As such, anyone using HTTP should be aware of this.
On the ESP, it is meant for private networks, where the Wifi provides the actual security.
Closing as non-issue.

closed 2017-10-10 by devyte