esp8266/Arduino #2674: [Question] Can I drop the scannetworks result?

joaohcca, 2016-11-08:
Hello all, I would like to know if it is possible to performance a scan network and don't get the feedback into the module I just need the AP's receive the request.  I looked for the ESP8266WiFiScan library and cpp file but the whole thing seems to happen on command line on the .cpp files

`(wifi_station_scan(&config, reinterpret_cast<scan_done_cb_t>(&ESP8266WiFiScanClass::_scanDone))`

Is there another file defining what wifi_station_scan(...) does? Or should I look to a lower level language? maybe Lua or try to use the Spressif sdk? 

If it helps I'm using Arduino 1.6.8 with a LoLin board (Nodemcu similar) with a ESP8266 12E.
with the following code
`#include <ESP8266WiFi.h>
void setup() {
  Serial.begin(9600); 
//  WiFi.mode(WIFI_STA);
//  WiFi.disconnect(); 
ESP8266WiFiScanClass wf;    
int n;
n = wf.scanNetworks(false,false);       
Serial.println("");                               //using n for debug, dont need him 
Serial.print("n = ");
Serial.println(n);                       
}
 void loop() {
 int SleepSec = 20; 
 ESP.deepSleep(SleepSec*1000000, WAKE_RF_DEFAULT); 
 delay(100);
  }`

Any suggestions would be great?

liquidfalcon, 2016-11-08:
The return value of scanNetworks is simply a count of the number of networks it found, so you can then get the information about them. It will perform the scan regardless of if you use the end result.

joaohcca, 2016-11-09:
I know, but I assume that the module process some feedback from the AP's and that's what a want to avoid. If possible I want the module to be 'visible' to AP's but remain the AP's 'invisible' to the module.

liquidfalcon, 2016-11-09:
WiFi scanning doesn't send anything to the access point - It simply tells the ESP to listen for beacon frames from all nearby broadcasting networks. All the ESP does is decode that frame and place it into a structure for you to pull the information from. The ESP never becomes 'visible' to the APs during this procedure, since it never sends anything.

devyte, 2017-07-12:
@igrr Considering the age of this, lack of response, and that it's a question rather than an issue, would you agree it can be closed?

closed 2017-07-13 by igrr