esp8266/Arduino #2675: PWM connected LEDs pulse when sending SPI commands to ili9341 LCD

horendus, 2016-11-09:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.1.0-rc2

### Description

I have connected an ili9341 2.8inch LCD display to a ESP8266 (NodeMCU Board) using SPI bus.
I am driving the LED Backlight with a PWM signal from pin16 which works well until I lower the PWM value down to about 50% to dim the screen.

On the LCD I have the TIME updating every second. Every time the screen updates to re draw the second (only that section is re drawn not whole screen) it also causes the backlight to pulse. It doesnt appear to do with when the screen is at high brightness level (PWM 1023).

My theory is that using the SPI bus causes disruption to the PWM signal. Is this a know issue?
Is there a work around? 

I plan to try this library out tonight to see if it makes a difference. 
https://github.com/StefanBruens/ESP8266_new_pwm

### Settings in IDE

Module:  			NodeMCU v1.0
Flash Size: 		4MB
CPU Frequency:		120Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL?
Reset Method:		nodemcu

(This isnt the real sketch as its over 8,000 lines over 10 .ino files but its just an example of how I am setting the PWM backlight control)
### Sketch

#include <Arduino.h>
int backlight = 16
void setup() {
pinMode(backlight ,OUTPUT);
digitalWrite(backlight, 400);
}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2016-11-11:
Have you tried another pin ?

horendus, 2016-11-15:
The work around I used was to switch to D16 for PWM signal because after checking with an oscilloscope it was the only pin which generated a stable signal without a 20mv flicker at the time of heavy SPI usage (such as screen draws)

I tried PWM with D0, D10 andD16 as they were the only pins I could make available in my project, the rest in use by various devices such as , RTC, ili9241 LCD, 2xRelays. 

Anyone have a theory to why D16 is the only stable PWM pin when using heavy SPI?

davidgevorgian, 2016-11-16:
I have the same problem last month.  Try connected via a diode and a capacitor to the ground, see picture.
[image]

davidgevorgian, 2016-11-16:
Look table below:
IO0 & IO10 used for SPI

[image]

HERE Espressif Recommendation for PWM: IO4,IO12,IO14,IO15

[image]

devyte, 2017-10-10:
@horendus the ESP has software PWM, not hardware. The most likely reason for the dimming is that interrupts get disabled and enabled during SPI comms, which can then affect the software PWM.
It's also possible that you have a power supply voltage dip, but in your case I think that's less likely.
Did you try out the other lib? Be aware that its licensing is different from that of this repo, so it can affect you if you intend to distribute the code. 
About GPIO16, it has a pulldown, whereas all other GPIOs have a pullup. Not sure how that could affect, though.
I find this very interesting, but this is not the right place to discuss. Closing per #3655 .

closed 2017-10-10 by devyte