esp8266/Arduino #2678: HTTP POST problem

GolPesar132, 2016-11-09:
I have have a problem to send somthing use POST method and then read the page. my code is below:

`       
        #include <Arduino.h>
        #include <ESP8266WiFi.h>
        #include <ESP8266WiFiMulti.h>
        #include <ESP8266HTTPClient.h>
       #define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

        HTTPClient http;

void setup() {

    USE_SERIAL.begin(115200);
   // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }
    WiFiMulti.addAP("SSID", "PASSWORD");
}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;

        USE_SERIAL.print("[HTTP] begin...\n");
        // configure traged server and url
        http.begin("http://server/test.php"); //HTTP

        http.addHeader("Content-Type", "application/x-www-form-urlencoded");
         http.POST("id=s1&api=123456789");
         http.writeToStream(&Serial);
         http.end();

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();

        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
    }

    delay(10000);
}
`
when i run it i get the 411 error code. whats wrong here? how can i solve it?

maurotrochez, 2016-11-10:
Hi @GolPesar132 . Do you test this request in [POSTMAN?](https://www.getpostman.com/) is for discard that the service in php is good or you missing any parameters.

GolPesar132, 2016-11-10:
No I didn't test in postman. Because i had a host for this. I didn't miss any parameter. I don't know what do you mean about "is for discard that the service in php is good".

GolPesar132, 2016-11-10:
I test it use POSTMAN and its OK. I don't know why it tell give this error. how can I solve it?

maurotrochez, 2016-11-10:
Try add another header.
`http.addHeader("Content-Length", 0);`
or
`String` payload = "id=s1&api=123456789"`
`http.addHeader("Content-Length", payload.length());`

GolPesar132, 2016-11-10:
I try both of them but these don't work again.

maurotrochez, 2016-11-10:
Try using WifiClient, like in this [comment](https://github.com/esp8266/Arduino/issues/1390#issuecomment-170343041) 👍

jamesmarlowe, 2016-11-21:
The relevant section of your code is right here:
```
    // configure traged server and url
    http.begin("http://server/test.php"); //HTTP

    http.addHeader("Content-Type", "application/x-www-form-urlencoded");
     http.POST("id=s1&api=123456789");
     http.writeToStream(&Serial);
     http.end();

    USE_SERIAL.print("[HTTP] GET...\n");
    // start connection and send HTTP header
    int httpCode = http.GET();
    ...
```
You set up your POST, then erroneously `http.writeToStream(&Serial);` and `http.end();` before going on to do a GET request and checking the status code of that.

To fix your code you will need to rethink what you are doing, start by removing your first `http.end()` and the `http.GET()`.

GolPesar132, 2016-11-21:
@jamesmarlowe Thanks for your guide.
I remove first `http.end();`  and change `http.POST("id=s1&api=123456789");` to ` int httpCode =http.POST("id=s1&api=123456789");` and remove `http.GET()`. It's wrong again. What should I do now?

jamesmarlowe, 2016-11-21:
No problem, it's impossible to say what's wrong without your environment, but perhaps a complete POST example will help you see what you are missing: https://github.com/esp8266/Arduino/blob/master/tests/device/test_http_client/test_http_client.ino#L52

devyte, 2017-10-10:
@GolPesar132 this is not the right place to ask for help with your particular code. This is an issue tracker, meant for tracking issues in the core code. If you still need help, please refer to a community forum.
Closing per #3655 .

closed 2017-10-10 by devyte