esp8266/Arduino #2679: undefined reference to `lwip_socket'

comebackgb, 2016-11-10:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

Can't compile my code if use lwIP functions.How to fix the issue?Help! I have defiend LWIP_SOCKET 1 in "lwipopts.h".

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

extern "C"
{
  #include <lwip/netdb.h>
  #include <lwip/sockets.h>
}

void setup()
{
  lwip_socket(PF_INET, SOCK_STREAM, 0);
}

void loop()
{

}

```

### Debug Messages

```
messages here
sketch/test8266.ino.cpp.o:(.text.setup+0x0): undefined reference to `lwip_socket'
sketch/test8266.ino.cpp.o: In function `setup':
/Users/XXX/Documents/Arduino/projects/test8266/test8266.ino:11: undefined reference to `lwip_socket'
collect2: error: ld returned 1 exit status
exit status 1

akaJes, 2017-01-12:
liblwip.a not compiled with LWIP_SOCKET and i dont know implemented sockets or not if even recompile library

copercini, 2017-02-17:
Same problem here =(

igrr, 2017-02-18:
Note that simply recompiling the library with LWIP_SOCKET isn't sufficient to provide socket APIs. You would also need to implement sys_thread_new, sys_mbox_new, and other sys_ functions which LwIP uses as an abstraction layer on top of the OS. This isn't a trivial exercise with the non-OS SDK. I'm going to close this because I don't have plans for working on such an implementation. If anyone is in working on this, feel free to comment here.

closed 2017-02-18 by igrr