esp8266/Arduino #2681: GETTING A LIST OF CONNECTED CLIENTS

sergiom968, 2016-11-11:
How I can get a list of devices connected to my esp8266 in Access Point mode, with MAC address and IP address?

leonyuhanov, 2016-11-17:
I use this:

```
struct station_info *stat_info;
struct ip_addr *IPaddress;
IPAddress address;

void dumpClients()
{
  Serial.print(" Clients:\r\n");
  stat_info = wifi_softap_get_station_info();
  while (stat_info != NULL)
  {
    IPaddress = &stat_info->ip;
    address = IPaddress->addr;
    Serial.print("\t");
    Serial.print(address);
    Serial.print("\r\n");
    stat_info = STAILQ_NEXT(stat_info, next);
  } 
}
```

sergiom968, 2016-11-17:
Thank you so much

leonyuhanov, 2016-11-23:
@sergiom968 You should close this ticket if its solved

mkeyno, 2016-11-24:
@leonyuhanov  what header file should be inculcated to avoid scope error  
`error: 'wifi_softap_get_station_info' was not declared in this scope`

leonyuhanov, 2016-11-24:
put this at the top:
```
extern "C" { 
#include<user_interface.h>
}
```

closed 2016-11-25 by sergiom968

franciscogimeno2000, 2019-03-07:
error: invalid conversion from 'ip4_addr*' to 'ip_addr*' [-fpermissive]
IPaddress = &stat_info->ip

Hi, when compiling, I get this error, someone knows
why?

Thank you

phisp, 2019-12-01:
It cannot convert from the "ip4_addr" type to the "ip_addr" type. There is no conversion function defined.
I resolved the problem with:
`ipv4_addr *IPaddress = &stat_info->ip;`
`address = IPaddress->addr;`