esp8266/Arduino #2682: i2s input - is it supported in the SDK?

mph070770, 2016-11-11:
I see that i2s.h from here: `Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266` supports adding a DAC to the ESP8266 - what about reading from an ADC. Is that supported? I'd like to be able to record audio on i2s and save it to SPI flash.

```
void i2s_begin();
void i2s_end();
void i2s_set_rate(uint32_t rate);//Sample Rate in Hz (ex 44100, 48000)
bool i2s_write_sample(uint32_t sample);//32bit sample with channels being upper and lower 16 bits (blocking when DMA is full)
bool i2s_write_sample_nb(uint32_t sample);//same as above but does not block when DMA is full and returns false instead
bool i2s_write_lr(int16_t left, int16_t right);//combines both channels and calls i2s_write_sample with the result
bool i2s_is_full();//returns true if DMA is full and can not take more bytes (overflow)
bool i2s_is_empty();//returns true if DMA is empty (underflow)
```

Thanks

MarcFinns, 2017-02-20:
Any news on this front?

devyte, 2017-10-10:
@mph070770 this is not the right place to ask such questions. This is an issue tracker, meant to track issues in the core. Please discuss at a forum like esp8266.com or stackoverflow. If there is a specific feature request, feel free to open a new issue for discussion here, and include details and concrete usage examples.
Closing per #3655 .

closed 2017-10-10 by devyte