esp8266/Arduino #2684: WiFiClientBasic example slow to connect

Craig1516, 2016-11-12:
### Basic Infos  Using NodeMCU Dev board in the Arduino IDE 
                           Attempting to run WiFiClientBasic example

#### Hardware
Hardware:	ESP-12E

### Description

Successfully loaded the example program and am obtaining simple data from a GET request.  My problem is that the first time through Loop takes either 7.9 or 3.9 seconds (and nothing in between).  The second time through loop takes less than 200 ms.  I'm hoping that the first time though can be as fast as the second time though.  The design attempt is to report a bus schedule at the push of a button, then go to deep sleep.   On recovery (RST) 7.9 seconds is a bit long to wait.  Anybody know how I can speed up the performance the first time around.

/*   WiFiClinetBasic from Arduino IDE example set
 *  This sketch sends a message to a TCP server
 *
 */

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
unsigned long TMserviceTime = millis();
String url;

ESP8266WiFiMulti WiFiMulti;

void setup() {
    Serial.begin(115200);
    delay(10);
    Serial.printf("Begin time %d \n",  millis() - TMserviceTime);
//    ESP.eraseConfig();
    
//    // We start by connecting to a WiFi network
//    WiFiMulti.addAP("ssid", "password");

//    Serial.println();
//    Serial.println();
//    Serial.print("Wait for WiFi... ");

//    while(WiFiMulti.run() != WL_CONNECTED) {
//        Serial.print(".");
//        Serial.printf("Connected      time %d \n",  millis() - TMserviceTime);
//        delay(500);
//    }

//    Serial.println("");
//    Serial.println("WiFi connected");
//    Serial.println("IP address: ");
//    Serial.println(WiFi.localIP());

//    delay(500);
}

void loop() {
    const uint16_t port = 80;
    const char * host = "trimet.org"; // ip or dns
//    const char * host = "192.168.1.1"; // ip or dns

    
    Serial.printf("Begin loop     time %d \n",  millis() - TMserviceTime);

    Serial.print("connecting to ");
    Serial.println(host);
    url = "/arrivals/tt_by_txt?locids=3800";

    // Use WiFiClient class to create TCP connections
    WiFiClient client;

    if (!client.connect(host, port)) {
        Serial.println("connection failed");
        Serial.println("wait 5 sec...");
        delay(5000);
        return;
    }

    Serial.printf("client.connect time %d \n",  millis() - TMserviceTime);

    // This will send the request to the server
    client.print(String("GET ") + "http://trimet.org/" + url + "\r\n" + "Connection: keep-alive\r\n\r\n");

    //read back one line from server
    String line = client.readStringUntil('\r');
    Serial.println(line);
    client.println(line);

    Serial.println("closing connection");
    client.stop();
    
    Serial.println("wait 5 sec...");
    delay(5000);
}

devyte, 2017-10-10:
@Craig1516 no basic info, no IDE settings.
I'm not aware of such a long startup time in latest git, which makes me suspect something related to DNS, but that's a WAG. 
If the issue is still valid, please test and report back here.

Craig1516, 2017-10-10:
Hi Develo,

 

Not sure what this relates to.  I’m a fan of esp8266’s but I’m not Craig1516.  I saw your online comment and received this email.  Hopefully Craig1516 is dialed in to the blog.  Hope you can reach him/her.  

 

Hope it helps,

Craig

devyte, 2017-10-10:
@Craig1516 weirdest thing to happen to me here in a long time. Maybe an email change or something. Sorry for the trouble!
Closing.

closed 2017-10-10 by devyte