esp8266/Arduino #2686: Arduino Time library doesn't work

robertgregor, 2016-11-13:
Hello,
it seems, that if I will include Time.h in the sketch, it will not include the Arduino Time library, but something else.
I was searching for Time.h and it looks like, that there is extenza time.h only...

pipi61, 2016-11-13:
Hi!
try: https://github.com/PaulStoffregen/Time  or time lib from arduino original site
if not work try rename files and include lines in source files
 or copy files to project folder
etc

robertgregor, 2016-11-13:
I see, yes, I can do this, but I have a feeling, that it should be a part distro...

oqibidipo, 2016-11-16:
Include TimeLib.h instead. Lowercase time.h is part of the standard C library, and that's what you are getting on a case insensitive file system.

devyte, 2017-10-10:
@robertgregor no sketch, no basic info, no IDE settings.
I am using TimeLib.h, together with an NTPClient lib, and I can't reproduce any issues with latest git.
Closing.

closed 2017-10-10 by devyte