esp8266/Arduino #2689: OTA TCP transfer fails

jbillhei, 2016-11-15:
With ESP-12F, core 2.3.0, BasicOTA example.

The mDNS discovery and TCP connection work fine, but then the TCP connection breaks.  See attached Wireshark capture.

The error status I get is OTA_END_ERROR.

The actual wifi connection is fine, and I can stream UDP all day with no issues.  OTA is the only thing I use TCP for, so I do not know if there is a problem with the OTA code or in lwip.

[image]

devyte, 2017-10-10:
@readmodifywrite is this issue still valid in latest git?

readmodifywrite, 2017-10-12:
Sorry, no idea. I ended up designing my own OTA system.  Feel free to close if no one else is having an issue.

closed 2018-07-12 by devyte