esp8266/Arduino #2690: 8266 Resetting and Locking Up...

dudesky71, 2016-11-16:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:		AdaFruit HUZZAH and WeMos D1 Mini Pro
Core Version:      	2.3.0

### Description

Problem description

### Settings in IDE

Module:  			AdaFruit HUZZAH and WeMos D1 Mini Pro
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		 SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

The sketch used is the Adafruit IO MQTT Example. I've also used "WiFiManager" as a separate test and the same thing happens which leads me to believe there is a core issue with the 8266 code.

```

### Debug Messages

```
I just tried another WeMos module. I've tried a HUZZAH. I've tried the basic MQTT client and they all lock up or reset the module within minutes of running. Please see the log and dump below... Can you help with this? 

Thank You

Sending counter val 248...OK!

Sending counter val 249...OK!

Sending counter val 250...OK!

Sending counter val 251...OK!

Sending counter val 252...OK!

Sending counter val 253...OK!

Sending counter val 254...OK!

Sending counter val 255...OK!

Sending counter val 256...OK!

Sending counter val 257...OK!

Sending counter val 258...Failed
Connecting to MQTT... Connection failed
Retrying MQTT connection in 5 seconds...
Connection failed
Retrying MQTT connection in 5 seconds...

MQTT Connected!

Sending counter val 259...OK!

Sending counter val 260...Failed
Connecting to MQTT... 
MQTT Connected!

Sending counter val 261...Failed
Connecting to MQTT... 
MQTT Connected!

Sending counter val 262...Failed
Connecting to MQTT... Connection failed
Retrying MQTT connection in 5 seconds...

MQTT Connected!

Sending counter val 263...OK!

Sending counter val 264...OK!
Connecting to MQTT... Connection failed
Retrying MQTT connection in 5 seconds...
Connection failed
Retrying MQTT connection in 5 seconds...
Connection failed
Retrying MQTT connection in 5 seconds...

Soft WDT reset

ctx: cont 
sp: 3ffef710 end: 3ffef900 offset: 01b0

stack>>>
3ffef8c0: 00000001 3ffee638 3ffee814 40201e32

3ffef8d0: 3fffdad0 00000000 3ffee8c8 40201e88

3ffef8e0: 3fffdad0 00000000 3ffee8c8 40203844

3ffef8f0: feefeffe feefeffe 3ffee8e0 40100718

<<<stack<<<
ets Jan 8 2013,rst cause:2, boot mode:(1,7)

ets Jan 8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

liquidfalcon, 2016-11-16:
How are you powering these boards?

dudesky71, 2016-11-16:
I have powered them from the USB port 5V from a PC and from a 5V 2A Wall adapter. I get the same results.

navivanuva, 2017-03-17:
Is this issue fixed? I have the same problem. ESP8266 runs fine for some time (around 6 hours), then the board locks up with the message:

ets Jan 8 2013,rst cause:4, boot mode:(1,7)

wdt reset

liquidfalcon, 2017-03-17:
@navivanuva - Which module are you using? When it "locks up", does using the reset button/pin get it running again, or does it remain in bootloader mode until you power cycle it?

navivanuva, 2017-03-18:
@liquidfalcon i am using esp8266-01 module. Yes, it runs again when i use the reset button.

devyte, 2017-10-10:
@dudesky71 you are essentially asking for help for an issue in a 3rd party lib. Assuming that the issue is still valid in latest git, I suggest asking the lib author for help with debugging. If the author can't/won't help, I suggest asking at a community forum. 
If an issue in the core libs in this repo is found, please open a new issue and provide a MCVE sketch, plus additional details.
Closing.

closed 2017-10-10 by devyte