esp8266/Arduino #2691: docs: staging version is currently behind stable

karlp, 2016-11-16:
### Basic Infos

#### Hardware
Hardware:			N/A
Core Version: master

### Description

The readme.md refers to "stable" v2.3.0, and also a "staging" version, http://arduino.esp8266.com/versions/2.3.0-rc2/package_esp8266com_index.json  the label for this is 2.3.0-rc2, but the contents of the json file refer to 2.3.0-rc1.  Both of these are behind current stable 2.3.0.

### Settings in IDE

N/A

### Sketch

N/A
### Debug Messages

N/A

igrr, 2016-11-17:
If you scroll the file down a bit, you'll find 2.3.0-rc2 in there.

karlp, 2016-11-17:
Ok, yes, the staging json file lists multiple packages.  I've edited the description.  The main point that "staging" is currently behind the stable release still stands.

devyte, 2017-10-10:
@igrr I have no idea how to check this, or how to tell what the current status is for latest git/2.4-rc1 etc. Could you please provide guidance? Can this be closed?

igrr, 2017-10-10:
Yes, this can be closed. "Staging" links have been removed, and current pre-release version (2.4.0-rc1) is ahead of the stable one.

closed 2017-10-10 by igrr