esp8266/Arduino #2693: ESP Tool will not connect to Adafruit Huzzah ESP8266

claudiaAl, 2016-11-16:
### Basic Infos

#### Hardware
Hardware:		Adafruit Huzzah ESP8266
Core Version:      	2.3.0

### Description
ESP tool will not successfully upload.  Connection is never made.  I have confirmed that I am in boot mode, and that I am uploading to the correct board.  Any thoughts?

### Settings in IDE

Module:  			Adafruit HUZZAH ESP8266
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:	nodemcu

### Sketch

```cpp

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>

const char ssid[] = "**********";
const char password[] = "***********";
const char host[] = "192.***.*.*";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(57600);
  Serial.setTimeout(10);
  delay(100);

  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

}

void loop() {
  // put your main code here, to run repeatedly:
  WiFiClient client;
  const int httpPort = 5000;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  String url = "http:// 192.***.*.*:5000/getID";
  String query = String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n";
  client.print(query);
  uint32_t timeout = millis() + 1000;
  while (client.available() == 0) {
    if (timeout < millis()) {
      Serial.println(">>> Client Timeout !");
      client.stop();
      return;
    }
  }

    while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  
  Serial.println();
  Serial.println("closing connection");
}
```

### Debug Messages

```
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbserial-A104JS84
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.usbserial-A104JS84 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
```

timlag1305, 2016-11-17:
What Operating System are you using? I believe that if you are using Linux you will need to add your user to specific groups that are mentioned here: https://wiki.archlinux.org/index.php/Arduino#Accessing_serial

I don't know if you have to do anything like that on other operation systems.

claudiaAl, 2016-11-18:
I'm on a Mac using Sierra 10.12.1.  And i just confirmed that I have R/W permissions on the port.  Any other ideas out there?  Thanks!

tablatronix, 2016-12-03:
Have you tried manual reset first? Maybe it is not actually resetting...

kendo55, 2016-12-04:
i had that problem too !!! 

FIX FOR USING 2.3.0 CORE (only .begin if not connected)!!!!!!!

if (WiFi.status() != WL_CONNECTED) {     // FIX FOR USING 2.3.0 CORE (only .begin if not connected)
    WiFi.begin(ssid, password); // connect to the network
}
while (WiFi.status() != WL_CONNECTED) {
    delay(500);
Serial.print(".");
}

see:
(#2186)
and:
(#2594)

devyte, 2017-10-10:
@claudiaAl is this issue still relevant? Is the programmin mode set correctly? (qio/dio)

devyte, 2018-08-01:
Closing due to lack of feedback in a long while.

closed 2018-08-01 by devyte