esp8266/Arduino #2697: espcomm_upload_mem failed on LoLin NodeMcu v3

pro100svitlo, 2016-11-18:
#### Hardware
Hardware:			[NodeMcu v3 (ESP-12E) LoLin](http://www.tdegypt.com/product/nodemcu-v3-lolin/)

### Description

I am using Arduino IDE 1.6.8 as described [here](https://github.com/esp8266/Arduino#installing-with-boards-manager)
- Install Arduino 1.6.8 from the Arduino website.
- Start Arduino and open Preferences window.
- Enter http://arduino.esp8266.com/stable/package_esp8266com_index.json into Additional Board
- Open Boards Manager from Tools > Board menu and install esp8266 platform 
Finish with this steps
When trying to run blink example get an error

### Settings in IDE
Module:  			Generic ESP8266 Module
FlashMode:         DIO
FlashFrequency: 40 Mhz
CpuFrequency:   80Mhz
FlashSize:            4M(3M SPIFFS)
DebugPort:          Serial
ResetMethod:      sk
UploadSpeed:     9600

### Sketch

```cpp
#include <Arduino.h>

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}
```

### Debug Messages

```
Sketch uses 222,157 bytes (21%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 31,524 bytes (38%) of dynamic memory, leaving 50,396 bytes for local variables. Maximum is 81,920 bytes.
Uploading 226304 bytes from /var/folders/c2/8ccg49m51r38v_6j85y96ghr0000gp/T/buildf584dae8d135c82c1bc9490aba96cc60.tmp/Blink.ino.bin to flash at 0x00000000
warning: espcomm_send_command: wrong direction/command: 0x01 0x08, expected 0x01 0x02
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
error: espcomm_upload_mem failed
```

saratma, 2017-02-17:
@pro100svitlo , i also get the same response every time i tried. did you got any solution for the same?
Anyway i keep trying to figure out to solve the problem. Hope someone may help us to get it done.

everslick, 2017-02-17:
i could not upload to NodeMcu V3 with esptool. as soon as i switched to esptool.py it worked. but i do not use the Arduino IDE which uses esptool for all I know.

JamesGKent, 2017-02-17:
might be a stupid question, but when you try to flash the device, are you holding down the flash button before applying power? might be worth trying to ensure the ESP is ready to receive a new sketch

Prophet19, 2017-04-23:
Eu consegui resolver meu problema instalando o driver correto e apertando o botão de Flash antes de ligar a minha placa.

kessedk, 2017-05-19:
I have the exact same problem, only I'm using Android IDE 1.8.2.

mrlynch71, 2017-05-20:
I just bought it today, and have the same problem, it never worked, always same error when uploading sketch. When I press upload at Arduino IDE, blue led at the NodeMcu V3 blink twice and the stop, 3 seconds latter Arduino shows the error.
At the begining, I had problems when tried to conect NodeMcu V3, port show an error and Control Panel. I load driver got from CH340 manufacturer´s web site. Then Device appeared OK, but I really don´t know if this could be the root cause of Arduino IDE´s error message

devyte, 2017-10-10:
@pro100svitlo reset method looks wrong, it should be Nodemcu instead of sk.
I have several of these exact same boards, and no issues while flashing.
Closing due to user error, and/or can't reproduce.

closed 2017-10-10 by devyte

Gettyk, 2018-11-27:
I had the same error, after trying to upload a new sketch while the previous one was an [OTA](https://www.fais-le-toi-meme.fr/en/electronic/tutorial/esp8266-arduinoota-sketch-upload-over-the-air). 

I solved this error by selecting the correct port in Arduino in the Tools menu (it should be set to set to a COM port not to an IP address). If you have several port, go in devices manager, open "Ports (COM &LPT) and note the numbre next to COM on the line "USB serial CH340). 

If you don't see any COM port in arduino or in devices manager, you should change your USB cable and get a data USB cable.

wouterds, 2019-01-26:
Same issue for me, using [LoLin NodeMCU V3](https://www.ebay.com/itm/153059049343).