esp8266/Arduino #2698: Feature: wifimulti persisent support

tablatronix, 2016-11-19:
Add sdk persistent support for wifimulti.

Some info from @kentaylor

kentaylor commented on May 10

> I looked into this a bit more. It turns out the device will store data for up to 5 WiFi access points. It will only use one of them rather than switch between them by itself but once one is selected it will handle the connectivity automatically. The Espressif SDK functions you would use are:- 
> 1. `wifi_station_ap_number_set` to set how many to save, might as well make it 5.
> 2. `wifi_station_get_ap_info` which returns details of the access points recorded and can be tested against the list of scanned access points.
> 3. `wifi_station_ap_change` to set which one to use.
> See function definitions in sedctions 3.4.10 - 3.4.12 at http://www.mikrocontroller.net/attachment/245197/2C-SDK-Espressif_IoT_SDK_Programming_Guide_v0.9.5.pdf .
> 
> The algorithm when you save one would be choose the next empty one to save to. If there is already 5 saved, delete number 1. Then move 2 to 1 etc to make a vacancy at 5 and store the new one at 5.
> At startup, scan for WiFi networks then change to the one from the scan list with the strongest signal that matches the SSID of one of the stored ones.

ref: https://github.com/tzapu/WiFiManager/issues/166

tablatronix, 2017-08-27:
Has this been considered or dealt with elsewhere?

a-c-sreedhar-reddy, 2019-02-22:
Who is working on this enhancement?

devyte, 2019-02-22:
Nobody atm, current focus is on stabilizing 2.5.0.

tavdog, 2019-06-22:
I would love this feature !

d-a-v, 2019-08-28:
Is anybody willing to work on this ?

a-c-sreedhar-reddy, 2019-08-29:
should the available wifi's be stored in FLASH?

devyte, 2019-08-29:
No, the request is for ::persistent(true|false) or. similar. The value would then be forwarded internally to WiFi.persistent(val). This is usually used  as false to reduce flash wear.

tablatronix, 2019-08-30:
Yes it is to make the wifimulti persistent, I have not tested the SDK functions above to test if this even works though.

Erriez, 2020-10-03:
FYI: Yesterday my WiFi Multi redesign is merged into master.

@devyte requested to look at this old request from @tablatronix and I made a proof of concept. Please find my results below:

The ESP8266 contains an array to store up to 5 WiFi configurations to flash. You're right: `wifi_station_ap_number_set()` must be called in order to change the default from 1 to 5.

After this change, new configurations can be saved to flash with `wifi_station_set_config()`. It looks like saving the configs in a ring, but I could not reliable save multiple SSID's after each other without a `system_restart()`. (It is vague in the datasheet how it works)

The ESP82866 does not automatically connect to the next registered network on connection failure. It must be called manually with  `wifi_station_ap_change(index);`.

However: **Saving WiFi credentials to flash contains a serious security issue.**

Consider the following use case:
- Store one or more SSID/Passwords in flash with `wifi_station_set_config()`.
- Keep the ESP8266 operating in your garden or somewhere else.
- A hacker steals your ESP8266 device.
- The hacker writes the following exploit to the ESP8266 and overwrites the sketch:

```c++
// Fixed 5 configs
#define NUM_CONFIGS 5

void printConfigs()
{
    struct station_config config[NUM_CONFIGS];
    
    // Clear config
    memset(config, 0, sizeof(config));

    // Print configs
    Serial.printf(PSTR("Get AP info: %s\n"), wifi_station_get_ap_info(config) ? "Ok" : "Failed");
    for (uint8_t i = 0; i < NUM_CONFIGS; i++) {
        Serial.printf(PSTR("Config %d:\n"), i);
        Serial.printf(PSTR("  ssid:      %s\n"), config[i].ssid);
        Serial.printf(PSTR("  password:  %s\n"), config[i].password);
        Serial.printf(PSTR("  bssid_set: %d\n"), config[i].bssid_set);
        Serial.printf(PSTR("  bssid:     %02x:%02x:%02x:%02x:%02x\n"), 
            config[i].bssid[0], config[i].bssid[1], config[i].bssid[2],
            config[i].bssid[3], config[i].bssid[4], config[i].bssid[5]);
        Serial.printf(PSTR("  threshold.rssi: %d\n"), config[i].threshold.rssi);
        Serial.printf(PSTR("  threshold.authmode: %d\n"), (uint8_t)config[i].threshold.authmode);
    }
}
```

Now all SSID's, passwords and MAC addresses are printed on the serial console in plain text which allows the hacker to come back and login to your WiFi networks... I know, security is a different topic, but I'm a little bit disappointed about this weak ESP8266 implementation.

I also noticed that this applicable in the current master version. For this reason, I think saving the WiFi credentials in plain text to flash is not the way forward... It would be more save to store SSID/password in encrypted flash with `WiFi.persistent(false);` to disable saving passwords this way.

What is your opinion?

Official API documentation:
https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf

Test sketch to experiment:

```c++
#include <ESP8266WiFi.h>
#include <PolledTimeout.h>

#define NUM_CONFIGS   5

#define WIFI1_SSID  ""
#define WIFI1_PASS  ""
const uint8_t WIFI1_BSSID[] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };

#define WIFI2_SSID  ""
#define WIFI2_PASS  ""
const uint8_t WIFI2_BSSID[] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };

[... 137 lines cut ...]
    if (wifi_station_ap_change(0)) {
        Serial.println(F("wifi_station_ap_change(): OK"));
    } else {
        Serial.println(F("wifi_station_ap_change(): Failed"));
    }
#endif

    printConfigs();
}

void loop() 
{
    if (wifi_station_get_connect_status() == 5) {
        Serial.printf(PSTR("Connected %s\n"), WiFi.SSID());
    } else {
        Serial.println(F("Disconnected"));
    }

    delay(1000);

    if (WiFi.status() != WL_CONNECTED) {
        connectNextAP();
    }
}
```

tablatronix, 2020-10-03:
>However: Saving WiFi credentials to flash contains a serious security issue.

This is unrelated, and it entirely doable right now simply by just dumping flash, but a good point to bring up in an esp sdk issue

Erriez, 2020-10-07:
@tablatronix and @devyte, Yesterday I dived deep into the SDK:

> It turns out the device will store data for up to 5 WiFi access points.

This is correct.

> It will only use one of them

Correct: The ESP8266 can use one of the registered WiFi AP configurations.

> rather than switch between them by itself 

- It turns out that the ESP8266 has no hardware built-in algorithm to automatically switch to an AP with strongest RSSI. This must be done by library/application code.

> once one is selected it will handle the connectivity automatically.

Partially correct:
- When the WiFi connection is lost, it always tries to reconnect to the same network, even when the AP is down.
- An automatic connect after power-on/reset or connection lost depends on the setting `wifi_station_set_auto_connect()`.
- Code is needed to: 1. detect a WiFi connection lost. 2. start a new WiFi scan 3. connect to the WiFi with best RSSI with WIFi config stored in RAM or flash. 
- The last connection will be used after power-on or reset when this is saved in flash and auto connect = 1.

Note: AP configurations can be erased with `system_restore()` which is similar to a factory reset:
-  wifi_station_set_auto_connect = 1
-  wifi_set_phy_mode = PHY_MODE_11N
-  wifi_softap_set_config -> All cleared with value 0
-  wifi_station_set_config -> All cleared with value 0
-  wifi_set_opmode -> SOFTAP_MODE = 2

> The algorithm when you save one would be choose the next empty one to save to. 

The SDK does not provide an interface to directly address the index of an AP config. It writes a new config incrementally to a new index: 0, 1, 2, 3, 4, wrapping to 0, etc.

Limitation: There is no SDK API to retrieve the number of AP's, set by `wifi_station_ap_number_set()`.
I cannot verify if calling `wifi_station_ap_number_set()` writes to flash every time to prevent flash wear-out.

Saving multi WiFi configs (max 5) to RAM works as expected (`wifi_station_set_config()`) and can be read all 5 configs with one call (`wifi_station_get_ap_info()`) before connecting to an AP.

**Problem:** I encountered a problem to read the saved WiFi configs (max 5) from flash after calling `wifi_station_set_config()`. 
- `wifi_station_get_ap_info()` returns all 5 WiFi configs stored in RAM and not from flash.
- `wifi_station_get_config_default() returns only the current WiFi config from flash and cannot be selected by an AP index 0..4.

The consequence is that the config index of the SSID/password is unknown after a `wifi_station_set_config()` call which is needed for the WiFi connect `wifi_station_ap_change(ap_index)`.

A workaround is to reset the CPU with `system_reset()` after the last `wifi_station_set_config()` call to read the updated configs from flash, but resetting is not a nice solution.

Test case for persistent to save to flash:
1. Erase flash once `system_restore(); system_restart();`
2. PHY wakeup: `wifi_fpm_do_wakeup(); wifi_fpm_close();`
3. Set station mode: `wifi_set_opmode(1);`
4. Register 1..5 AP's: `wifi_station_set_config(&config);`
5. Read all 5 AP configs: `wifi_station_get_ap_info();`
Only the last `wifi_station_set_config()` is displayed at index 0 which is incorrect. **(Bug in SDK?)**
`wifi_station_get_config()` cannot select index.
6. Reset CPU and point 5 displays correct configs from flash.
7. Run WiFi scan
8. WiFi connect with ap_index 0..4 `wifi_station_ap_change(ap_index);`

Due too this limitation, I'm not sure how to proceed.

tablatronix, 2020-10-07:
Wow you've done a great job with this, very thorough

Erriez, 2020-10-07:
Thanks! Now finding a solution. :-)

Abedi98, 2021-02-09:
> Thanks! Now finding a solution. :-)

Hi . excellent
Please explain how to use your code?
How to save access point information?
Please! Write a real example to use the code you generated

Thanks

Abedi98, 2021-02-09:
> Thanks! Now finding a solution. :-)
```

void rootPage() {
    String content = R"(
    <!DOCTYPE html>
    <html>
    <head>
      <meta charset="UTF-8" name="viewport" content="width=device-width, initial-scale=1">
    </head>
    <body>
    <b> Tik-Box Configuration </b>
    <form method="POST" action="/postForm">
      <input type="text" name="ssid" value="Write your ssid...">
      <input type="text" name="password" value="Write your wifi password...">
      <input type="submit" value="Save">
    </form>    
    </body>
    </html>
    )";

    server.send(200, "text/html", content);
}

void handlePostForm()
{
      String qsid = server.arg("ssid");
      String qpass = server.arg("password");
      String content;
      int statusCode;
  if(qsid != "" && qpass != ""){
      setCfg("newssid", "1234567890", 0);
      server.send(200, "text/html", "saced");
  }else{
    server.send(200, "text/html", "please fill all input");
  }
}
```
To save the access point, I wrote the following code. But when a new access point is saved, the previous access point is lost

Erriez, 2021-02-09:
@Abedi98 The easiest way is to use the WiFiMulti example in master:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiMulti/WiFiMulti.ino

Abedi98, 2021-02-09:
> @Abedi98 The easiest way is to use the WiFiMulti example in master:
> https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiMulti/WiFiMulti.ino

I do not want to do this as hard code in setup.
I want to save the ssids and be able to manage them on a separate page

like this :
```
void rootPage() {
    String content = R"(
    <!DOCTYPE html>
    <html>
    <head>
      <meta charset="UTF-8" name="viewport" content="width=device-width, initial-scale=1">
    </head>
    <body>
    <b> Tik-Box Configuration </b>
    <form method="POST" action="/postForm">
      <input type="text" name="ssid" value="Write your ssid...">
      <input type="text" name="password" value="Write your wifi password...">
      <input type="submit" value="Save">
    </form>    
    </body>
    </html>
    )";

    server.send(200, "text/html", content);
}

void handlePostForm()
{
      String qsid = server.arg("ssid");
      String qpass = server.arg("password");
      String content;
      int statusCode;
  if(qsid != "" && qpass != ""){
      setCfg("newssid", "1234567890", 0);
      server.send(200, "text/html", "saced");
  }else{
    server.send(200, "text/html", "please fill all input");
  }
}
```

d-a-v, 2021-03-06:
OK to close this feature request per @Erriez's #7619 and #7629 ?

Erriez, 2021-03-07:
@d-a-v, yes you can close the feature requests.

closed 2021-03-07 by d-a-v