esp8266/Arduino #2699: [Linux] Compile Error - fatal error: WifiServer.h: No such file or directory

SciLor, 2016-11-19:
See
https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/issues/11
https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/issues/5

Linux specific (Ubuntu)

If you use WifiServer.h in a cpp Header you get following error:
```
In file included from ~/testordner/ESP8266_Hyperion_LED-Controller/HyperionRGB/HyperionRGB.ino:14:0:
sketch/WrapperJsonServer.h:6:24: fatal error: WifiServer.h: No such file or directory
#include <WifiServer.h>
```

WrapperJsonServer.h:
https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/blob/master/HyperionRGB/WrapperJsonServer.h

igrr, 2016-11-20:
> SciLor/ESP8266_Hyperion_LED-Controller#11
> SciLor/ESP8266_Hyperion_LED-Controller#5

Which Arduino IDE version are you using?
Have you included ESP8266WiFi.h from the .ino file?

SciLor, 2016-11-20:
I am using IDE 1.6.12
I have just added
```
#include <WifiServer.h>
#include <WifiClient.h>
```
Into my Header file where I am using the WifiServer.h

PS: It works fine under Windows,

igrr, 2016-11-20:
The file is called WiFiServer.h, not WifiServer.h.

SciLor, 2016-11-20:
That could be the problem thanks. Linux distinguishes between small and large letters. I am going to test it out.