esp8266/Arduino #2700: Active tcp closing socket

akaJes, 2016-11-19:
how can I send from ESP active closing command? because stop() not working
as described https://en.wikipedia.org/wiki/Transmission_Control_Protocol#Connection_termination
[image]

akaJes, 2016-11-22:
after code digging I have one question "why You destroy tcp_pcb after sending `tcp_close()`?"
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/src/include#L69-L89
after `tcp_close()` socket(pcb) sends FIN_WAIT_1 and must wait acknowledge from another side for proper closing and cleanup it after some timeout
like it is doing `espconn` interface

devyte, 2017-10-10:
@akaJes is this issue still relevant in latest git? If so, please provide a MCVE sketch, as well as any relevant additional details.

devyte, 2017-12-05:
Closing due to no feedback in almost 2 months, and due to age.

closed 2017-12-05 by devyte