esp8266/Arduino #2702: No WiFi connect since 2.3.0 if delay >= 1000 in setup() for init of external hardware

heiko-wolf, 2016-11-21:
#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0

### Description
No WiFi connect since 2.3.0 if "delay >= 1000" in setup() for init of external hardware.
After upgrading to 2.3.0 many sketches wont connect anymore to WiFi Accesspoint.
Testing a few things showed up that the problem was caused by adding a delay(1000) in setup before WiFi.begin().
This was done to power up external hardware properly before starting the Arduino/ESP Code and connecting the WiFi. 
After downgrading to 2.2.0 everything works fine again using this delay.

### Settings in IDE

Module:  			Wemos D1 R2 & mini
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz

### Sketch

#include <ESP8266WiFi.h>
char ssid[50] = "your ssid";
char password[50] = "your pwd";

void setup() {
  Serial.begin(9600);

  delay(1000); // for ext. hardware init  <<<<<<<<<<<

  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  delay(1000);
  Serial.println("X");
}

kendo55, 2016-11-30:
i had that problem too !!! The Problem is not the `delay().....`

FIX FOR USING 2.3.0 CORE (only .begin if not connected)!!!!!!!

```
if (WiFi.status() != WL_CONNECTED) {     // FIX FOR USING 2.3.0 CORE (only .begin if not connected)
    WiFi.begin(ssid, password); // connect to the network
}
while (WiFi.status() != WL_CONNECTED) {
    delay(500);
Serial.print(".");
}
```

see: 
(https://github.com/esp8266/Arduino/issues/2186) 
and:
(https://github.com/esp8266/Arduino/issues/2594)

heiko-wolf, 2016-11-30:
Hi there,
youre right, after adding a WiFi.mode(WIFI_OFF) at the beginning of setup() and enable it just directly before WiFi.begin() after all other stuff is done and a few delays later averything works fine again :)
Thanks for pointing to the other issues !!

closed 2016-11-30 by heiko-wolf

cimba007, 2017-01-29:
Got the same problem but none of the workarounds seems to fix it.
I do quite some heavy init stuff (8 Seconds) and put WiFi.mode(WIFI_OFF); at the beginning of setup.

After my 8 seconds I do this

` 
 WiFi.mode(WIFI_STA);
  listNetworks(0);
  if (WiFi.status() != WL_CONNECTED) {  // FIX FOR USING 2.3.0 CORE (only .begin if not connected)
      WiFi.begin(ssid, password);       // connect to the network
  } 
 `
Still sometimes I got no networks found and after that no connection ..

mlord, 2017-02-27:
Try using static IP rather than DHCP -- fixes it for me here.  The internal DHCP seems to have issues under some conditions.

johnwa27, 2017-05-04:
I just discovered this wonderful feature 😡.  The workaround suggested in kendo55's post begs the question, "why is the module connected prior to WiFi.begin?".  The workaround could be problematic if there is more than 1 AP and you need to connect to one specific one or change from one to the other.
I added a WiFi.disconnect() before WiFi.begin().  So far it has worked and I think it is a more robust solution.

ratnick, 2018-10-24:
I think [kendo55](https://github.com/kendo55) nailed it. It seems like the wifi.begin() needs some time to fetch the response.
What he did in https://github.com/esp8266/Arduino/issues/2186 was:

```
if (WiFi.status() != WL_CONNECTED) {  // FIX FOR USING 2.3.0 CORE (only .begin if not connected)
    WiFi.begin(ssid, password);       // connect to the network
}    

for ( count = 0; count < 20 ; count++ )  {
      delay(500);
      wifi_stat = WiFi.status();  
      Serial.print( count ); Serial.print( " WifiStat: " );
      Serial.println( wifi_stat );
      if ( wifi_stat == WL_CONNECTED ) {
        break;
      }
}

Serial.println("");

if ( count < 20 ) {
    Serial.println("WiFi connected");
}else{
    Serial.println("Client !!!NOT!!! connected");

   // create a SoftAP for access
   WiFi.softAP(ESPssid, ESPpassword);
   // etc..........
}

```

... which worked for me (whereas all the other solutions above did not work in my case.