esp8266/Arduino #2703: AVRISP verification error; content mismatch (WINDOWS)

SirRibeiro, 2016-11-21:
Hello,

So i'm trying to program an arduino uno, with ESP12E in windows, using the AVRISP example.

In ubuntu using the following command:
avrdude -c avrisp -p atmega328p -P net:172.16.1.118:328 -U flash:w:one.ino.standard.hex

Everything works fine.
But trying this in windows is not working. 

In windows i'm using NetBurner Virtual COM port, to create a virtual port, to send the data over TCP to the ESP12E, the data is send, and some code is written, but in the end, gives a verification error; content mismatch, as you can see above.

If anyone can help me it would be great!
Thanks in advance.

#### Hardware
Hardware:		ESP-12E
NodeMCU 1.0 

### Description

AVRISP verification error; content mismatch (WINDOWS)

### Sketch

Arduino_Wifi_AVRISP

### Debug Messages

avrdude: Version 6.3, compiled on Sep 12 2016 at 17:24:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "D:\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM10
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.99s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\code\AppData\Local\Temp\arduino_build_860169/Blink.ino.hex"
avrdude: writing flash (928 bytes):

Writing | ################################################## | 100% 8.36s

avrdude: 928 bytes of flash written
avrdude: verifying flash memory against C:\Users\code\AppData\Local\Temp\arduino_build_860169/Blink.ino.hex:
avrdude: load data flash data from input file C:\Users\code\AppData\Local\Temp\arduino_build_860169/Blink.ino.hex:
avrdude: input file C:\Users\code\AppData\Local\Temp\arduino_build_860169/Blink.ino.hex contains 928 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 8.26s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0042
         0x00 != 0x15
avrdude: verification error; content mismatch

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected

FernandoGarcia, 2016-12-23:
Hi!
Try using AVRDUDESS I had this error with Windows too.
See my tutorial: https://www.youtube.com/watch?v=Ll35jVvetA8
Best regards.

devyte, 2017-10-10:
@SirRibeiro I don't understand the issue. Is it still relevant? If so, please provide more details (wiring, sketches, versions, etc).

devyte, 2018-08-01:
Closing due to lack of feedback in a long while.

closed 2018-08-01 by devyte