esp8266/Arduino #2706: OTA from Linux: no response from device, connect failed

tomtobback, 2016-11-22:
hi, i'm trying to get OTA to work on Linux with the BasicOTA sketch as per instructions. the OTA port shows up, but when i upload i get the 'no response from device' error in the IDE, and the Serial (via screen command) says:
Start
Progress: 0%
Error[2]: Connect Failed
Error[4]: End Failed

Configurations:
Ubuntu (crouton on Chromebook), IDE 1.6.8, python 2.7, NodeMCUv1.0
i also tried on another Ubuntu machine, another router, with IDE 1.6.12, python 2.7, NodeMCUv1.0 and same problem. latest ESP8266 package 2.3.0.
i find this command in the IDE:
python /home/tom/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/espota.py -i 192.168.100.179 -p 8266 --auth= -f /tmp/build8245f06855e5d3f6794a7bd0551b7de1.tmp/BasicOTA.ino.bin 
does that look OK? when i run it in an xterm, it returns the same error: no response from device
thanks for any hints- Tom

nomanshahsyed, 2017-03-22:
I have the same problem, I got the esp32 working with OTA but not esp8266. It gives the same errors. Have you already solved your problem?

tomtobback, 2017-03-22:
i have not tried again, was hoping to get some suggestions here

nomanshahsyed, 2017-03-22:
Well I figured it out just after commenting, seems as if the windows firewall was blocking the python. Unblocking did the trick. Maybe same might be the issue in linux.

devyte, 2017-10-10:
Per last comment, this is working on Windows, so it is likely an issue in the Ubuntu firewall. 
Closing as non-issue in core.

closed 2017-10-10 by devyte

klavatron, 2020-03-18:
Facepalm! Linux Mint 19. Switching firewall off works for me.