esp8266/Arduino #2709: transparent serial forward

ramarro123, 2016-11-22:
Hi,

looking at 
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino

i have try to use avrdude to upload a sketch on the chained arduino connected on serial.

i got some results, the arduino get correctly idenfitied, and some command pass, but  i got some timeout as well, that break the comunication.

i have try to write byte by byte, results are better, but still got some timeout that prevent the upload

someone can point me into the right direction for that?

penfold42, 2016-11-25:
This doesn't address your question but might be an alternative:
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266AVRISP

ramarro123, 2016-11-26:
umm, i have an 01 module, i don't see spi pin header, but i can be wrong on that

devyte, 2017-10-10:
@ramarro123 did you get this working?
ESP8266AVRISP should serve as a guideline, don't take it literally. In other words, I think it should work, as long as you adapt it to your case (e.g.: change the pins, adapt voltage, etc).
Having said that, this is not the right place to ask for assistance. Closing as non-issue, see #3655 .

closed 2017-10-10 by devyte