esp8266/Arduino #2710: Serial.available() and Serial.read() issue

TThanhXuan, 2016-11-23:
Hi everyone, I'm trying to use ESP8266 to read a string from keyboard and then print to Serial monitor. This is the code:
void setup() {
 Serial.begin(9600);
}

void loop() {
 if (Serial.available()>0) {
  String r = Serial.readStringUntil('\r');
  Serial.print("r: ");
  Serial.println(r);
 }
}

There is nothing happen when I run the code, nothing on Serial monitor.

Could you please help me with this issue ?
Thank you very much.

tablatronix, 2016-12-03:
Are you sure your client is sending \r, try using \n

devyte, 2017-10-10:
@TThanhXuan I suggest testing by reading and echoing char by char instead of reading until '\r'. Also, should that really be '\r' or 'n'?
In any case, this is not the right place to ask for assistance. Closing, see #3655 .

closed 2017-10-10 by devyte