esp8266/Arduino #2714: Connecting power RGB leds with MOSFET

esp8266india, 2016-11-25:
Hello, i'm trying to connect power LED strip to GPIO pins 12,13 and 04 on the ESP12E but the code is too complicated for me to write a lighthandler virtual function...is there any simpler way....
right now, hue android apps can detect it but how to do the real thing? My final aim is to connect esp based bridge to IFTTT.com so that the leds can be programmed automatically and respond to various conditions without myself doing anything.

penfold42, 2016-11-25:
I don't understand your question.

For the low level stuff just use analogwrite to use pwm to set the led brightness and a simple mosfet led driver circuit from google.

esp8266india, 2016-11-26:
Sorry i wrote that in a very complicated way.

I've a RGB LED strip with 60 leds. However it is not neopixel compatible. Rather, it has 4 wires, white-positive, Red, Green and blue for negative of those three colours respectively. I've managed to get colour of my choice using esp8266 creating a webserver. This is done by using 3 n-MOSFETS connected to Red, green and blue wires and to ground. Their gate is driven by 3 GPIO pins of the esp8266 using pulse width modulation technique as there is no true analog output from esp8266

But due to lack of knowledge of internet protocols, ssdp and all that, i am unable to follow the example code you gave for lighting the neopixel. To get colour of my choice i only need red, green and blue 
intensity levels separately as integer values from 0-255. 

If i am correct, there is a virtual function in your library for that *lighthandler, but i am unable to get red, green and blue levels there...any help is greatly appreciated. 

If i am successful, i will put these strips at strategic locations to get lighting effect according to time of day, music being played and what not. The reason i am trying for hue protocol is that there are so many apps already there to achieve these goals, i only have to make my led strip appear like a hue light bulb. I suppose, that will not be very difficult courtesy to your library as neopixel example you gave achieves this.

Another advantage of using philips hue protocol is that it can be directly connected to the cloud and using services like ifttt.com (please see: https://ifttt.com/hue for examples, if interested) i can very easily achieve something like the following (just a few examples, but very useful at least for me):
1. fading alarm--light should gradually increase intensity 6.00am-8.00am in morning much like the real thing.
2. alert lights -- if someone is outside my room, the PIR sensor gives a signal and the light should automatically turn somewhat more reddish in colour and increase intensity.
3. fitness tracker alert -- if i do not run according to my goal, the light should turn somewhat more yellowish
4. sleep assistant -- light shold gradually fade according to time and turn orangish in evening time.

example like point 3 which depend on different services can be achieved due to support of 3rd party...

i hope, it is more clear now about what i want to achieve...

esp8266india, 2016-11-26:
Extremely sorry, posted in wrong place.

closed 2016-11-26 by esp8266india