esp8266/Arduino #2717: ESP8266WiFiType.h is referencing queue.h.....where is the said file??

pioneerhemantjn, 2016-11-28:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:

ESP8266WiFiType.h is referencing queue.h.....where is the said file?? Thanks!! 

Can I get a downloadable zip of the entire workable library?

Many Thanks!!

#### Hardware
Hardware:			ESP-12E

me-no-dev, 2016-11-28:
https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/queue.h

pioneerhemantjn, 2016-11-28:
Thanks!!

The library has a problem with "ESP8266WiFiScan.h" file, getting the following error during code compile:-

\src/ESP8266WiFiScan.h:63:16: error: 'function' in namespace 'std' does not name a type

         static std::function<void(int)> _onComplete;

Can you share a completely _error-free_ library for use with ThingSpeak IoT!

Thanks!!

MahiraECE, 2017-03-17:
The same problem i also get 
Please select the Board ESP8266 in boards manager

rkkoszewski, 2017-09-24:
How to make this work on an Arduino Uno??

devyte, 2017-10-10:
@pioneerhemantjn No sketch, no basic info, no IDE settings.
I'm not aware of any compile issues in 2.3.0 or latest git.
If you need a .zip, you can get it from the github main repo page. Better yet, install via git, per instructions on readthedocs.
Closing due to can't reproduce.

closed 2017-10-10 by devyte

Goseale, 2020-06-19:
Find a way to replicate it, i tried many ways of installing this library and had the same issue, and when i supply the queue.h file another error comes out

d-a-v, 2020-06-19:
@Goseale give details on your installation ?
Did you select an esp8266-based board in the board manager ?
I just checked and `#include <queue.h>` in `ESP8266WiFiType.h` seems to be unused and removable. Can you try to comment it ?