esp8266/Arduino #2719: C:\Users\admin\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27:22: fatal error: functional: No such file or directory  #include <functional>

pioneerhemantjn, 2016-11-29:
Kindly provide a workable, rightly-structured library which can be used with UNO!!

Why such spurious includes are there in the concerned library files? Next is on "memory" - "include <memory>"
.
.
.
.

I suppose this is a matured product by now providing wifi connectivity, then there has to be a stable library too that can be used readily.

Please help someone!

THanks!!

vicnevicne, 2016-11-29:
I'm not sure what you mean, but if you intend to get Wifi on an Arduino using this project, you're at the wrong place.
This project (esp8266/Arduino) is meant to program and use an esp8266 with Arduino software (IDE etc). It is *not* meant to connect an esp8266 to Arduino hardware.
If you want to use the est8266 as a "slave" to an Arduino, there are several pages about that, such as http://shin-ajaran.blogspot.be/2014/12/noobs-guide-to-esp8266-with-arduino.html ...

pioneerhemantjn, 2016-11-30:
> be/2014/12/noobs-guide-to-esp8266-with-arduino.html ...

*Gentlemen,*

All I am wanting to do is to perform certain 'http' communications using
ESP8266-12E with an UNO with a whole lot of sites on the net.

Therefore all I am looking for is a *stable, workable C++ ESP8266 library*
to be used in my Arduino sketch which enables me to communicate over *http
with sites *on the net.

Incidentally the *SSID/AP-level* connection has been established days ago
and my *ESP/UNO* combo is online. Next one would simply like to do some
*http* conversations, bi-way, with certain *sites* on the net offering* IoT*
services, that's it!!

I just want a piece of *updated, most recent, stable,
structured ESP8266WiFi library* with a *webclient* and a *webserver*
example thrown in!!

What I have found and have been helped with gives errors on compilation,
including the contained examples!

I am not into programming of ESP8266/related stuff, just doing some*
IoT *streaming/
triggering using Arduino!

DO you have the library with all associated/referenced files at one place
to be used in an Arduino sketch.

Trust you can help.

*Thanks & Regards,*

*Hemant*

devyte, 2017-10-10:
@pioneerhemantjn If you want code for the UNO to use the ESP as a wifi bridge, then you're in the wrong repo.
If you want code to run directly on the ESP, then this is the right repo, but this is not the right place to request help for your particular project. Please ask at a community forum.
Closing as off-topic, see #3655 .

closed 2017-10-10 by devyte