esp8266/Arduino #2720: Scan IP address of ESP8266

anujmattoo, 2016-11-29:
I am working on a project in which, the mobile app/browser sends HTTP command to ESP12F in AP mode. The default IP address of ESP is 192.168.4.1. Therefore in order to connect the ESP to WiFi, the command goes like this :- 192.168.4.1/SSID=XYZ/Pass=XYZ.

On connecting to router, the ESP goes into STA mode and Doesn't create AP until WiFi is not found to which it was connected earlier. Now when ESP is connected to router, the router assigns the IP to ESP through which HTTP command will be sent. On Serial monitor, it does show the IP address to which it is assigned, but what if app controls the ESP, there is no serial communication. How will app know the IP address assigned to particular ESP12F? Is there any way app can scan the router? Or any other way to do this? If there are multiple ESP connected to router, how can I identify the IP of particular ESP.

penfold42, 2016-11-29:
Use some sort of service discovery or name resolution.

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/README.md

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266SSDP/README.md

devyte, 2017-10-06:
@anujmattoo previous comment is correct.
Closing due to off-topic, see #3655 .

closed 2017-10-06 by devyte