esp8266/Arduino #2721: SPIFFS on Sonoff from ITEAD

chrigu1981, 2016-11-30:
Hello,

any idea why SPIFFS is not working on Sonoff Device from ITEAD and i'm even not able to get it work on a standart ESP-12E, works fine on NodeMCU and Wemos D1 mini

after 
```
SPIFFS.begin()
```

got reset and never comes up again
```
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

Thanks for your help

vicnevicne, 2016-11-30:
Please note that Sonoff does not use a AI Thinker module. Their PCB includes the ESP8266 as well as 1MB of flash, so please make sure that is selected in the IDE. 
Some people have flashed it successfully, see https://tzapu.com/sonoff-firmware-boilerplate-tutorial/
Others have even upgraded it to 32MB, but it's more complex :-) : http://tech.scargill.net/32mb-esp01/

chrigu1981, 2016-12-01:
hello,

thx for your. answer. i already flashed a lot of Sonoff devices but my problem is, that SPIFFS is not working. Is there any pin modification required for this? i can even not use SPIFFS on a standart ESP-12E. Only Wemos and NodeMCU is working with SPIFFS

vicnevicne, 2016-12-01:
ESP-12E ? Strange. I've used SPIFFS on ESP-12E with no issue. I first uploaded a spiffs image with ESP8266FS before accessing it from code though, following the procedure on https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#uploading-files-to-file-system
Did you try that ? Did it succeed or did you get an error ?

devyte, 2017-10-06:
@chrigu1981 No basic info, no IDE settings.
I can vouch for SPIFFS working correctly in latest git on 12-E/F. 
If this issue is still valid, I suggest requesting assistance at a forum. If you track a problem to the core, please open a new issue and add details.
Closing due to lack of info, and can't reproduce.

closed 2017-10-06 by devyte