esp8266/Arduino #2723: ESP crashes after multiple http requests

WilbertE, 2016-12-02:
Hi Guys

I have an ESP8266-01 module and my program crashes after time. I think its a memory issue as it happens after a dozen or so http requests.
The problem according to the stack trace is in umm_malloc.c line 1733.

That is this function:
`void umm_free( void *ptr ) {

  ptr = GET_UNPOISONED(ptr);

  /* check poison of each blocks, if poisoning is enabled */
  if (!CHECK_POISON_ALL_BLOCKS()) {
    return;
  }

  /* check full integrity of the heap, if this check is enabled */
  if (!INTEGRITY_CHECK()) {
    return;
  }

  _umm_free( ptr );
}`

**This is the exception**
```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 11 results
0x4010068c: free at C:\Users\Wilbert\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x40208c0e: String::lastIndexOf(String const&, unsigned int) const at C:\Users\Wilbert\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4020211f: getPage(String) at C:\Users\Wilbert\Documents\Arduino\WiFi\temp\WiFi_ESP8266-01/WiFi_ESP8266-01.ino line 37
0x4010053d: _umm_realloc at C:\Users\Wilbert\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at C:\Users\Wilbert\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x40208a44: String::equals(String const&) const at C:\Users\Wilbert\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40208a93: String::equals(char const*) const at C:\Users\Wilbert\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40208ac5: String::startsWith(String const&, unsigned int) const at C:\Users\Wilbert\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x402023c0: _GLOBAL__sub_I__ZN16ESP8266WiFiClass9printDiagER5Print at ESP8266WiFi.cpp line ?
0x40208158: IPAddress::toString() at C:\Users\Wilbert\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/IPAddress.cpp line 87
0x40100718: cont_norm at C:\Users\Wilbert\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
```

And this is my code:
`//Function to send data to the website and get a response
//date should be in this format: [www.host.ext;/path/page.aspx?querystring;]
String getPage(String data){
  String segments[2];
  int lastIndex = 0;
  int counter = 0;
  

  //split the data into host and path
  for (int i = 0; i < data.length(); i++) {
    if (data.substring(i, i+1) == ";") {
      segments[counter] = data.substring(lastIndex, i);
      lastIndex = i + 1;
      counter++;
    }
  }
  const char* host     = segments[0].c_str();
  String path          = segments[1] + "&token="+ token;

  //Try to download the page
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return "";
  }

  client.print(String("GET ") + path + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");

  //Wait for server to respond
  unsigned long timeout = millis();
  while (client.available() == 0) {
	  if (millis() - timeout > 5000) {
		  Serial.println("timeout");
		  client.stop();
		  return;
	  }
  }

  //If there is data we return it
  int dir = 0; 
  String response = "";
  while(client.available()){
    String line = client.readStringUntil('\r');
    response += line;
  }
  client.stop();
  return response;
}`

WilbertE, 2016-12-02:
Ok, my  bad. It happens when I send 2 url's almost instantly.

devyte, 2017-10-06:
Closing due to last comment.

closed 2017-10-06 by devyte